v923z/micropython-ulab #156: Compile for LoBo ESP32 MicroPython

mdaeron, 2020-07-28:
I googled a bit but couldn't find any reports of successfully compiling [LoBo MicroPython for ESP32](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki) with ulab. I'm planning to try as soon as I receive the [TinyPICO](https://www.tinypico.com) I ordered recently, but at least one potential [problem](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/263) has been reported.

Does anybody have experience to share?

teuler, 2020-07-28:
I haven't recently tried the LoBo code, but the TinyPICO port works well with ulab. 
Here are my [notes](https://github.com/teuler/robotling/wiki/Adding-native-modules-to-MicroPython-(ESP32)#-adding-ulab-to-the-micropython-firmware).

v923z, 2020-07-28:
@teuler Thomas, can you compile with the new code? We massively re-organised in the last couple of days. I think it shouldn't make much of a difference, though...

teuler, 2020-07-28:
I compiled it on June 27 for the last time. I can test this likely on the weekend.

v923z, 2020-07-28:
> I compiled it on June 27 for the last time. I can test this likely on the weekend.

These are really recent changes (end of last week) that I am talking about.

mdaeron, 2020-07-28:
> I haven't recently tried the LoBo code, but the TinyPICO port works well with ulab.

That's good to know, thank you. By "TinyPICO port", do you mean the ESP32 port, or is there a specific TinyPICO port in mainline MicroPython?

Context: I need both `ulab` and `DAC.write_timed()` on a TinyPICO. AFAIK only the LoBo port offers the latter on ESP32.

teuler, 2020-07-29:
I was talking about the main ESP32 port of the official Micropython with the TinyPICO as target. I never used the DAC, therefore I don't know if it contains the functionality you need.

teuler, 2020-07-29:
After pulling the current versions of `micropython`, `micropython-lib` and `ulab`, I recompiled the standard Micropython EPS32 port targeted at the TinyPICO using the following makefile:

```
ESPIDF = $(HOME)/esp/esp-idf  
BOARD = TINYPICO  
PORT = /dev/ttyUSB0  
FLASH_MODE = dio  
FLASH_SIZE = 4MB  
#CROSS_COMPILE = xtensa-esp32-elf-  
include Makefile  
```
With the exception of two warnings (see below) it compiles and links fine. I haven't done any thorough checks but my current project, which uses `ulab` seems to run w/o obvious errors on the TinyPICO.

Here is the segment of the compiler/linker output that contains the warnings:
```
CXX /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp
In file included from /home/tom/esp/esp-idf/components/freertos/include/freertos/portable.h:89:0,
                 from /home/tom/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /home/tom/esp/esp-idf/components/cxx/cxx_guards.cpp:22:
/home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h: In function 'void cpu_configure_region_protection()':
/home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h:77:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < sizeof(pages_to_protect)/sizeof(pages_to_protect[0]); ++i) {
                       ^
CXX /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_encr.cpp
CXX /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp
CXX /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_storage.cpp
CXX /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp
CC /home/tom/esp/esp-idf/components/smartconfig_ack/smartconfig_ack.c
CC /home/tom/esp/esp-idf/components/spi_flash/flash_mmap.c
CC /home/tom/esp/esp-idf/components/spi_flash/partition.c
CC /home/tom/esp/esp-idf/components/spi_flash/spi_flash_rom_patch.c
CC /home/tom/esp/esp-idf/components/spi_flash/cache_utils.c
CC /home/tom/esp/esp-idf/components/spi_flash/flash_ops.c
CC /home/tom/esp/esp-idf/components/ulp/ulp.c
CC /home/tom/esp/esp-idf/components/ulp/ulp_macro.c
CC /home/tom/esp/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/netbuf.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/api_lib.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/netifapi.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/netdb.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/err.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/api_msg.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/api/sockets.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/apps/sntp/sntp.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/stats.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/mem.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/inet_chksum.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/init.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/memp.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/tcp_in.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/dns.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/ip.c
In file included from /home/tom/esp/esp-idf/components/freertos/include/freertos/portable.h:89:0,
                 from /home/tom/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_platform.hpp:19,
                 from /home/tom/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp:18:
/home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h: In function 'void cpu_configure_region_protection()':
/home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h:77:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int i = 0; i < sizeof(pages_to_protect)/sizeof(pages_to_protect[0]); ++i) {
                       ^
CC /home/tom/esp/esp-idf/components/lwip/lwip/src/core/raw.c
```

teuler, 2020-07-29:
RE DAC in this firmware:
```
>>> import machine
>>> dir(machine)
['__class__', '__name__', 'ADC', 'DAC', 'DEEPSLEEP', 'DEEPSLEEP_RESET', 'EXT0_WAKE', 'EXT1_WAKE', 'HARD_RESET', 'I2C', 'PIN_WAKE', 'PWM', 'PWRON_RESET', 'Pin', 'RTC', 'SDCard', 'SLEEP', 'SOFT_RESET', 'SPI', 'Signal', 'TIMER_WAKE', 'TOUCHPAD_WAKE', 'Timer', 'TouchPad', 'UART', 'ULP_WAKE', 'WDT', 'WDT_RESET', 'deepsleep', 'disable_irq', 'enable_irq', 'freq', 'idle', 'lightsleep', 'mem16', 'mem32', 'mem8', 'reset', 'reset_cause', 'sleep', 'soft_reset', 'time_pulse_us', 'unique_id', 'wake_reason']
>>> dir(machine.DAC)
['__class__', '__name__', 'write', '__bases__', '__dict__']
```

v923z, 2020-07-29:
> After pulling the current versions of `micropython`, `micropython-lib` and `ulab`, I recompiled the standard Micropython EPS32 port targeted at the TinyPICO using the following makefile...

Thanks for the feedback!

> ```
> /home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h: In function 'void cpu_configure_region_protection()':
> /home/tom/esp/esp-idf/components/soc/esp32/include/soc/cpu.h:77:23: warning: [...]

This seems to come from some ESP32 component, right? If there are warnings related to `ulab`, let me know, and I will try to sort it out.

v923z, 2020-07-29:
@teuler By the way, we removed the `extras` module; there is one `user`, which contains a dummy function, nothing more. You can move your `playground` there. Since we are not going to change anything in `user`, it should be easy to just pull `ulab`.

teuler, 2020-07-29:
RE warnings: yes you're right; these warnings are unrelated to 'ulab'; I just wanted to give the complete picture

v923z, 2020-07-29:
OK, thanks!

v923z, 2020-09-21:
@mdaeron I don't know, whether this is still pertinent, or we can close it.

On a different note, at one point you mentioned that you would be willing to help with testing on a macintosh. I have re-written `ulab` from scratch, and I was wondering, whether you could check out the https://github.com/v923z/micropython-ulab/tree/tensor branch, and try to compile it. Any fixes should be submitted against this branch. Thanks!

v923z, 2020-10-28:
This issue seems to be stale, so closing now.

closed 2020-10-28 by v923z