v923z/micropython-ulab #157: Some ulab functions have limit of uint16_t (0-65535) for size of vectors

rcolistete, 2020-07-29:
Testing ulab on MicroPython boards with some MBytes (ESP32 4 MB PSRAM, etc), it is easy to observe that some ulab functions have limit of uint16_t (0-65535) for size of vectors :
- array.reshape;
- approx.bisect/fmin/newton;
- fft.*;
- linalg.linspace;
- poly.polyfit;
- etc.

It think it is a matter of replacing 'uint' / 'uint16_t' for 'size_t'.

v923z, 2020-07-29:
> It think it is a matter of replacing 'uint' / 'uint16_t' with 'size_t'.

That's right. I will fix it later today.

rcolistete, 2020-07-29:
> > It think it is a matter of replacing 'uint' / 'uint16_t' with 'size_t'.
> 
> That's right. I will fix it later today.

It is important to revise all ulab functions using int/uint/int16/uint16 to confirm which ones have the 65535 limitation. My list above is problable not complete.

closed 2020-07-29 by v923z

v923z, 2020-07-29:
>     * array.reshape;

Done.

> 
>     * approx.bisect/fmin/newton;

This mostly concerns `maxiter`, and I don't think it would be wise to set that to `size_t`.

> 
>     * fft.*;
> 

Done.

>     * linalg.linspace;

`linspace` is a top-level function, and can be found in `ulab_create.c`. But I fixed it all the same.

>     * poly.polyfit;

I changed it, as far as the length of the array is concerned, but left it as it was for the degree of the polynomial.

> 
>     * etc.
> 

This is a bit vague ;o)

> It is important to revise all ulab functions using int/uint/int16/uint16 to confirm which ones have the 65535 limitation. My list above is probably not complete.

Well, I agree with the statement, but my resources are also limited. I usually fix issues within days, most of the time within hours, but the issues have to be reported first. 

In any case, if you find other function, where this might be a problem, just open a new issue.