v923z/micropython-ulab #161: Compilation of unix port on MacOS

mdaeron, 2020-07-31:
I used to be able to compile unix micropython with ulab following the instructions found [here](https://github.com/v923z/micropython-ulab#compiling), but somethings goes wrong now. Any idea what's happening here?

Below are the commands I type followed by the output of each command. The error is at the bottom.

+ `git clone https://github.com/v923z/micropython-ulab.git ulab`

```
Cloning into 'ulab'...
remote: Enumerating objects: 96, done.
remote: Counting objects: 100% (96/96), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 2694 (delta 45), reused 66 (delta 35), pack-reused 2598
Receiving objects: 100% (2694/2694), 1.55 MiB | 142.00 KiB/s, done.
Resolving deltas: 100% (1754/1754), done.
```

+ `git clone -b master https://github.com/micropython/micropython.git`

```
Cloning into 'micropython'...
remote: Enumerating objects: 81521, done.
remote: Total 81521 (delta 0), reused 0 (delta 0), pack-reused 81521
Receiving objects: 100% (81521/81521), 44.71 MiB | 117.00 KiB/s, done.
Resolving deltas: 100% (59071/59071), done.
```

+ `cd micropython; git submodule update --init`

```
Submodule 'lib/asf4' (https://github.com/adafruit/asf4) registered for path 'lib/asf4'
Submodule 'lib/axtls' (https://github.com/pfalcon/axtls) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/btstack' (https://github.com/bluekitchen/btstack.git) registered for path 'lib/btstack'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lwip' (https://git.savannah.gnu.org/r/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'lib/mbedtls'
Submodule 'lib/mynewt-nimble' (https://github.com/apache/mynewt-nimble.git) registered for path 'lib/mynewt-nimble'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path 'lib/nrfx'
Submodule 'lib/nxp_driver' (https://github.com/hathach/nxp_driver.git) registered for path 'lib/nxp_driver'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Submodule 'lib/tinyusb' (https://github.com/hathach/tinyusb) registered for path 'lib/tinyusb'
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/asf4'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/axtls'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/berkeley-db-1.xx'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/btstack'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/libffi'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/lwip'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/mbedtls'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/mynewt-nimble'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/nrfx'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/nxp_driver'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/stm32lib'...
Cloning into '/Users/mathieu/temp/ulab/micropython/lib/tinyusb'...
Submodule path 'lib/asf4': checked out 'd270f79aa16dd8fd4ae3b6c14544283dcb992e9c'
Submodule path 'lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
Submodule path 'lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
Submodule path 'lib/btstack': checked out 'c8b9823f68c6af0fa52e2c4e009aba4dbf257232'
Submodule path 'lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Submodule path 'lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'lib/mbedtls': checked out '3f8d78411a26e833db18d9fbde0e2f0baeda87f0'
Submodule path 'lib/mynewt-nimble': checked out '97ce3eacaaa79e8ed6cf71717149ced4f5328ee7'
Submodule path 'lib/nrfx': checked out '7a4c9d946cf1801771fc180acdbf7b878f270093'
Submodule path 'lib/nxp_driver': checked out 'b618cb1d521cc9e133bdcd0fca154dee2d925dfe'
Submodule path 'lib/stm32lib': checked out '58fee7c92bd576814d3f2afd92fbc62990270ecc'
Submodule path 'lib/tinyusb': checked out 'a6b916ba85bef6aad50f1652532b02984dfe2484'
```

+ `cd mpy-cross; make`

```
[... 369 lines cut ...]

+ `make CFLAGS='-Wparentheses-equality' USER_C_MODULES=../../../ulab all`

But with `ulab` this fails with the following output:

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../ulab/code
mkdir -p build-standard/code
mkdir -p build-standard/code/approx
mkdir -p build-standard/code/compare
mkdir -p build-standard/code/fft
mkdir -p build-standard/code/filter
mkdir -p build-standard/code/linalg
mkdir -p build-standard/code/numerical
mkdir -p build-standard/code/poly
mkdir -p build-standard/code/user
mkdir -p build-standard/code/vector
CC ../../../ulab/code/ndarray.c
../../../ulab/code/ndarray.c:17:10: fatal error: 'py/runtime.h' file not found
#include "py/runtime.h"
         ^~~~~~~~~~~~~~
1 error generated.
make: *** [build-standard/code/ndarray.o] Error 1
```

v923z, 2020-07-31:
Can you roll back to find out which commit brought it this error? I am groping in the dark here, since I'm using linux...

mdaeron, 2020-07-31:
> Can you roll back to find out which commit brought in this error? I am groping in the dark here, since I'm using linux...

I'll try that as soon as I have a better network connection.

I think this should be as simple as running `git checkout <commit hash>` in the `ulab` directory before compiling. Does this sound right ? My git skills are limited.

v923z, 2020-08-01:
> > Can you roll back to find out which commit brought in this error? I am groping in the dark here, since I'm using linux...
> 
> I'll try that as soon as I have a better network connection.
> 

You have a complete local copy, so you don't need network. 

> I think this should be as simple as running `git checkout <commit hash>` in the `ulab` directory before compiling. Does this sound right ? 

That's correct.

mdaeron, 2020-08-04:
My bad, this was a mistake on my part. For future reference, here is a script which works on MacOS Catalina:

```sh
# brew install homebrew/cask/gcc-arm-embedded
# brew install libffi
# brew install pkgconfig

export PKG_CONFIG_PATH=/usr/local/opt/libffi/lib/pkgconfig
export LDFLAGS="-L/usr/local/opt/libffi/lib"
export CPPFLAGS="-I/usr/local/opt/libffi/include"

echo "\n--- CLONE ULAB ---\n"
git clone https://github.com/v923z/micropython-ulab.git ulab

echo "\n--- CLONE MICROPYTHON ---\n"
git clone -b master --depth 1 https://github.com/micropython/micropython.git

echo "\n--- GIT SUBMODULE UPDATE ---\n"
cd micropython
git submodule update --init

echo "\n--- MAKE MPY-CROSS ---\n"
cd mpy-cross
make

echo "\n--- MAKE UNIX ---\n"
cd ../ports/unix
echo "\nCFLAGS += -Wno-parentheses-equality" >> Makefile
make USER_C_MODULES=../../../ulab all
```

mdaeron, 2020-08-04:
I just created this [PR](https://github.com/v923z/micropython-ulab/pull/166) to get rid of the 4 instances of double parentheses which were causing errors when compiling for unix on MacOS.

v923z, 2020-08-04:
Many thanks, I merged your changes.

closed 2020-08-04 by v923z