v923z/micropython-ulab #162: pi, degrees, radians

teuler, 2020-08-01:
Hi,
maybe I overlooked it, but it would be nice to have pi, and for code clarity also degrees() and radians() in ulab. The idea is that ideally, one would import `ulab` for all math instead of additionally having to import `math`.
Thanks

v923z, 2020-08-01:
I haven't included these, because, as you point out, they are either in the standard math module, or are trivial. On the other hand, I see your point.

v923z, 2020-08-01:
@teuler https://github.com/v923z/micropython-ulab/pull/163 implements the requested changes. The PR adds approx. 600 bytes to the firmware. I am a bit uncertain as to whether it is worth it for things that are already available. 

@jepler Before merging, I would like to hear your opinion.

teuler, 2020-08-01:
@v923z Thanks for the quick response. I guess it comes down to the question how often 'ulab' can fully replace importing  'math'; then again, 'math' is already in the firmware.
I tend to say that the added memory load is not worth the convenience of not having to import 'math'.

teuler, 2020-08-01:
Background of my question: I have code that runs both in MicroPython and normal Python (w/ 'numpy'). Using 'import ulab as np' or 'import numpy as np' allows more elegant code :)

v923z, 2020-08-01:
> Background of my question: I have code that runs both in MicroPython and normal Python (w/ 'numpy'). Using 'import ulab as np' or 'import numpy as np' allows more elegant code :)

That could be, but `ulab` is split into sub-modules, so, e.g., if you want to evaluate a polynomial, then you have to 

```
from ulab import poly

poly.polyval(p, x)
```
which under `numpy` is simply 

```
import numpy as np

np.polyval(p, x)
```

We defined sub-modules, so that partial firmware can be compiled for smaller systems: https://micropython-ulab.readthedocs.io/en/latest/ulab.html#customising-ulab If you are willing to forgo the modularity (i.e., want to compile all modules, or you don't care, if your code throws an error at runtime), I could probably do this as in `numpy`, and you could then set a pre-processor switch for `numpy`-like or `ulab`-like syntax. The problem is not really the code, but the documentation. It could become a tad confusing.

v923z, 2020-10-30:
This, and a lot of other features have been added in version 1.0.0. Closing the issue now.

closed 2020-10-30 by v923z