v923z/micropython-ulab #168: Implement `numpy.array.round()`

mdaeron, 2020-08-14:
It would be useful to have an efficient way to round a `ndarray`, equivalent to `numpy.array.round()`, either returning a new rounded `ndarray` or rounding the original `ndarray` in place.

v923z, 2020-08-14:
Is https://micropython-ulab.readthedocs.io/en/latest/ulab.html#around not what you need?

mdaeron, 2020-08-14:
Duh. Yes, it is. I searched `round` in [readthedocs](https://micropython-ulab.readthedocs.io/en/latest/index.html) and it did not return that function. Thanks.

closed 2020-08-14 by mdaeron

v923z, 2020-08-14:
@mdaeron 
> Duh. Yes, it is. 

No problem.

> I searched `round` in [readthedocs](https://micropython-ulab.readthedocs.io/en/latest/index.html) and it did not return that function.

`numpy` itself is a bit ambiguous: https://numpy.org/doc/stable/reference/generated/numpy.ndarray.round.html