v923z/micropython-ulab #169: `vector.floor()` and `vector.ceil()` return ndarrays of dtype `float`

mdaeron, 2020-08-14:
Both `numpy.floor()` and `numpy.ceil()` return integer ndarrays, but this is not the case in ulab. Is there a strong reason this is not the case? When computing a waveform to feed a `DAC()` or `I2S()` instance, I find myself doing awkward things in ulab, e.g.:

```
array(vector.sin(linspace(0, pi))*255, dtype = int16)
```

v923z, 2020-08-14:
> Both `numpy.floor()` and `numpy.ceil()` return integer ndarrays, 

I think this is not true. Output from `numpy`:

```
a = array([1, 2, 3], dtype=uint8)
b = ceil(a)
c = floor(a)
print(a, a.dtype)
print(b, b.dtype)
print(c, c.dtype)
```
```
[1 2 3] uint8
[1. 2. 3.] float16
[1. 2. 3.] float16
```

mdaeron, 2020-08-14:
> I think this is not true.

Duh again. I seem to be spectacularly wrong tonight. I tried earlier, but somehow reached the wrong conclusion. I'll refrain from opening new issues for a while, until my brain reboots :) Thanks.

closed 2020-08-14 by mdaeron

v923z, 2020-08-14:
> > I think this is not true.
> 
> I'll refrain from opening new issues for a while, until my brain reboots :) 

Don't worry, if there is an issue, it should be brought up.