v923z/micropython-ulab #171: Explicit `dtype` in functions

mdaeron, 2020-08-17:
This is a feature request/suggestion. I don't know how strictly we want to "map" `ulab` function signatures to `numpy` function signatures, but from a user perspective, it would be handy to let functions returning ndarrays specify a `dtype` argument, so that one could simply code:

```py
# two nested functions:
a = ulab.vector.floor(
	ulab.linspace(0,100)**2,
	dtype = ulab.int16)
```

instead of:

```py
# three nested functions:
a = ulab.array(
	ulab.vector.floor(ulab.linspace(0,100)**2),
	dtype = ulab.int16)
```

I agree that it's subjective whether this is simpler to write when coding, but I'm wondering whether at the implementation level there is some possible performance gain if you tell such functions (here, `ulab.vector.floor`) in advance that the result to return should be of a certain type?

If there are actually no performance gains, or if this sounds like too much woork for too little gain, of if we don't want to deviate from `numpy`, feel free to forget it.

v923z, 2020-08-17:
> This is a feature request/suggestion. I don't know how strictly we want to "map" `ulab` function signatures to `numpy` function signatures, 

As closely as possible. The following throws an exception in `numpy`:

```
a = array([4, 2, 3], dtype=uint8)
sin(a, dtype=uint8)
```

> but from a user perspective, it would be handy to let functions returning ndarrays specify a `dtype` argument, so that one could simply code:
> 
> ```python
> # two nested functions:
> a = ulab.vector.floor(
> 	ulab.linspace(0,100)**2,
> 	dtype = ulab.int16)
> ```
> 

I think, in most cases this wouldn't make too much sense. E.g., 

```
sin(linspace(...), dtype=ulab.int16)
```
would just return a number of zeros, right, while the intent was obvious different? If we do this, the user really has to know what they are doing.

> I agree that it's subjective whether this is simpler to write when coding, but I'm wondering whether at the implementation level there is some possible performance gain if you tell such functions (here, `ulab.vector.floor`) in advance that the result to return should be of a certain type?

That is really the main problem: it would add too much to the firmware size. This could be done in two possible ways.

Simple: call the `array` constructor internally, which would mean that the intermediate results (the float array) would have to sit in RAM, so then it would really just be a shortcut for your three nested functions. 

Complicated: when substituting into the function, we would have to catch the results before they are inserted into the output array. This would mean that the loop that computes would have to implemented for each output data type. I think that is expensive in terms of flash.

> If there are actually no performance gains, or if this sounds like too much woork for too little gain, of if we don't want to deviate from `numpy`, feel free to forget it.

My strongest argument against including this is the incompatibility with `numpy`.

mdaeron, 2020-08-17:
OK, it makes sense.

closed 2020-08-17 by mdaeron