v923z/micropython-ulab #176: Is it possible build ulab on double FP?

water5, 2020-08-26:
I try to build ulab with double FP using 
`make BOARD=PYBV11 MICROPY_FLOAT_IMPL=double USER_C_MODULES=../../../ulab all`
but not success.

v923z, 2020-08-26:
But is this a `ulab` issue? I mean, what happens, if you drop the `USER_C_MODULES` variable?

And what do you mean by `no success`? How does failure manifest itself?

water5, 2020-08-27:
When I build without USER_C_MODULES , the build process pass,
if included USER_C_MODULES, appear error below:

> ...
...
CC ../../../ulab/code/ulab.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC ../../lib/littlefs/lfs2.c
CC ../../lib/littlefs/lfs2_util.c
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
arm-none-eabi-ld: build-PYBV11/code/vector/vectorise.o: in function `vectorise_around':
vectorise.c:(.text.vectorise_around+0x84): undefined reference to `round'
make: *** [Makefile:605: build-PYBV11/firmware.elf] Error 1

and I refer [https://github.com/micropython/micropython/issues/4380](url), modify /ports/stm32/mpconfigport.h
`#define MICROPY_FLOAT_IMPL          (MICROPY_FLOAT_IMPL_FLOAT)`
to
`#define MICROPY_FLOAT_IMPL                  (MICROPY_FLOAT_IMPL_DOUBLE)`

Above problem exist still.

I try to disable 'vector' module via modify micropython-ulab/code/ulab.h , delete:
`// vectorise (all functions) takes approx. 6 kB of flash space`
`#ifndef ULAB_VECTORISE_MODULE`
`#define ULAB_VECTORISE_MODULE (1)`
`#endif`

The build process pass.

Additional, I try to build double FP with ulab (included vector module) on PYBD_SF6 (it has double-precision hardware floating point), build process pass, although I have no PYBD_SF6 board.

v923z, 2020-08-27:
> CC ../../../ulab/code/ulab.c
> CC ../../lib/oofatfs/ff.c
> CC ../../lib/oofatfs/ffunicode.c
> CC ../../lib/littlefs/lfs2.c
> CC ../../lib/littlefs/lfs2_util.c
> CC build-PYBV11/pins_PYBV11.c
> LINK [...]

OK, this brings us a bit closer to the solution. Apparently, `round` or `roundf` is not included in the compilation process. As a workaround, if you don't need the `around` function, you could comment that out the following lines:

https://github.com/v923z/micropython-ulab/blob/5c75f08118dc871bd3b6c143db6fbbd5ce4d1651/code/vector/vectorise.c#L111-L141

https://github.com/v923z/micropython-ulab/blob/5c75f08118dc871bd3b6c143db6fbbd5ce4d1651/code/vector/vectorise.c#L463

I believe, if you do that, it should pass. Obviously, this is only a beauty plaster, and we have to find out what the underlying problem is. 

> I try to disable 'vector' module via modify micropython-ulab/code/ulab.h , delete:
> `// vectorise (all functions) takes approx. 6 kB of flash space`
> `#ifndef ULAB_VECTORISE_MODULE`
> `#define ULAB_VECTORISE_MODULE (1)`
> `#endif`
> 

Actually, you don't have to delete anything, you can set the value to 0:
```
#ifndef ULAB_VECTORISE_MODULE
#define ULAB_VECTORISE_MODULE (0)
#endif
```

> 
> Additional, I try to build double FP with ulab (included vector module) on PYBD_SF6 (it has double-precision hardware floating point), build process pass, although I have no PYBD_SF6 board.

This again seems to indicate that one of the header files has an error. I will try to figure out, which one it is. If it is indeed a header file, then we'll have to raise the issue at `micropython`.

v923z, 2020-08-27:
@water5 If I run 
```
make BOARD=PYBV11 MICROPY_FLOAT_IMPL=double USER_C_MODULES=../../../ulab all

```
I get the firmware:

```
CC build-PYBV11/pins_PYBV11.c
LINK build-PYBV11/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 419380	    124	  27244	 446748	  6d11c	build-PYBV11/firmware.elf
GEN build-PYBV11/firmware.dfu
GEN build-PYBV11/firmware.hex
```

Did you pull the latest `micropython` source? This issue was actually fixed in https://github.com/micropython/micropython/pull/6242.

water5, 2020-08-28:
> (And it is some 15 kB larger than the version with single precision.) Did you pull the latest `micropython` source? I believe, this issue was actually fixed in [micropython/micropython#6242](https://github.com/micropython/micropython/pull/6242).

Thanks for your answer, pull latest MicroPython source, build pass.

closed 2020-08-28 by water5