v923z/micropython-ulab #177: Omega Arithmetic algorithm ?

pidou46, 2020-09-11:
Accelerometer are easily available on mcu, it would be handy in many case to be able to get speed and even position from these acceleration signal.
This involve double integration of the signal, classic integration method does not give good result but "Omega Arithmetic" algorithm seems to give much better results.
Do you think it could be added to ulab ?

[http://prosig.com/wp-content/uploads/pdf/blogArticles/OmegaArithmetic.pdf](http://prosig.com/wp-content/uploads/pdf/blogArticles/OmegaArithmetic.pdf)

[https://stackoverflow.com/questions/28106323/integration-in-fourier-or-time-domain](https://stackoverflow.com/questions/28106323/integration-in-fourier-or-time-domain)

v923z, 2020-09-11:
Your second example doesn't use anything beyond what is already available in `ulab`, right? You basically want a wrapper for that, I think. But if it so, wouldn't it be better to write the wrapper in python?

v923z, 2020-11-11:
@pidou46 I think I would close the issue, because the integration can be done using `trapz` in the `approx` sub-module. I have improved its accuracy in a recent commit.

closed 2020-11-11 by v923z