v923z/micropython-ulab #178: Feature request: numpy.concatenate

mouryarahul, 2020-10-05:
Hi,
I look through the documentation but didn't find any function to join/concatenate two or more arrays. Is there any workaround for such operation in current ULAB? Will you consider adding such a function in the future? 

Cheers,
Rahul

v923z, 2020-10-05:
I am really close to wrapping up the `tensor` branch https://github.com/v923z/micropython-ulab/tree/tensor, which has a number of new features, like support for broadcasting, views, and tensor rank up to 4, and is highly customisable via pre-processor constants https://github.com/v923z/micropython-ulab/blob/tensor/code/ulab.h. 

This latter statement also implies that we would no longer be bound by firmware size, because you could really easily exclude functions, if you don't need them, so adding extra features like what you ask for would definitely be doable. I definitely wanted to add `concatenate`. 

If you are adventurous, you can definitely check the branch out. Most things should work, but at the moment, there is no documentation. There is a small glitch with the subscription operator, otherwise, the code is done. 

As a workaround, you could do the following:

```python

import ulab

a = ulab.array([1, 2, 3, 4])
b = ulab.array([1, 2, 3, 4])
c = ulab.array(range(len(a) + len(b))
c[:len(a)] = a
c[len(a):] = b
```

mouryarahul, 2020-10-05:
Thanks for the advice. Tensor up to rank 4 will be really cool and usable in many applications.

v923z, 2020-10-05:
> Thanks for the advice. Tensor up to rank 4 will be really cool and usable in many applications.

And it eats a lot of flash space;) With all currently included functions, 4 dimensions require around 100 kB of flash.

v923z, 2020-10-13:
@mouryarahul Rahul, I have just looked at what is required for `concatenate`, and it seems to me that `numpy` upcasts the arrays. That is quite expensive. Based on my experience with the binary operators, I estimate `concatenate` to cost something between 2 and 3 kB for the four-dimensional case. I find that a bit too much for a single function. 

Would it be OK, if we imposed a condition on the `dtype`? E.g., arrays can be concatenated, if their `dtype`s are identical. If we do that, we gain a factor of about 20 in flash space.

The other possibility is that, instead of writing out the loops for all possible combinations, we implement a single loop, and delegate the type conversion to a function. That would not save so much space as the previous option, and it would also be slower, because you would have to dispatch a function call within the loop. However, for the price it would be universal.

Last question: do you want `concatenate` to be able to deal with higher-dimensional tensors, or just the linear case?

mouryarahul, 2020-10-13:
@v923z to save the Flash memory, I think 2nd option seems to be viable. If a user wants to concatenate two tensors, he/she can first convert one of the tensors compatible with the other in python. This will serve most of the purposes.

v923z, 2020-10-14:
> @v923z to save the Flash memory, I think 2nd option seems to be viable. If a user wants to concatenate two tensors, he/she can first convert one of the tensors compatible with the other in python. This will serve most of the purposes.

But that is the first option, isn't it?

mouryarahul, 2020-10-24:
Yeah, I think we can close this issue.

closed 2020-10-24 by mouryarahul