v923z/micropython-ulab #179: Request: circular buffer

roger-, 2020-10-11:
Originally raised [here](https://github.com/v923z/micropython-ulab/issues/14):

> How about a circular buffer compatible with ndarray (but fixed size of course)? Would avoid all the memmoves needed with roll().

v923z, 2020-10-12:
I am not against the idea, but we first should iron out how the buffer is supposed to work. 

1. Should this be a property of the `ndarray`, or should it be a sub-class? Sub-classing is difficult at the C level, while making it a property of the `ndarray` results in a `numpy`-incompatible structure.
2. What is supposed to happen, if you call `argmax(circular_array)`?
3. Should the circular buffer be supported for 1D arrays only, or for the general case?
4. What does `roll` do, when faced with a circular buffer?

Apart from the behavioral issues mentioned above, there is also the implementation issue. If you want to loop through an array at the C level, you can simply do something like 

```c
uint8_t *array = (uint8_t *)ndarray->array;
for(size_t i=0; i < ndarray->len; i++) {
   // do something with the value of the array
   result = some_function(*array);
   array += ndarray->strides[ULAB_MAX_DIMS - 1]; 
}
```

If your array doesn't start at the null position of the pointer, shifting the pointer becomes non-trivial. You could, perhaps, do something like 

```c
uint8_t *array = (uint8_t *)ndarray->array;
for(size_t i=0; i < positon; i++) {
   // do something with the value of the array
   result = some_function(*array);
   array += ndarray->strides[ULAB_MAX_DIMS - 1]; 
}
array -= ndarray->strides[ULAB_MAX_DIMS - 1] * ndarray->shape[ULAB_MAX_DIMS - 1]; 
for(size_t i=position; i < ndarray->len; i++) {
   // do something with the value of the array
   result = some_function(*array);
   array += ndarray->strides[ULAB_MAX_DIMS - 1]; 
}
```

But the first question is, what the actual use case would be. I understand that you would like to save the `memcpy` operation of `roll`, but `memcpy` itself is not very expensive.

street-grease-coder, 2020-10-28:
I might be wrong here, but I think most user needs would be satisified by implementing a queue function as exists in python https://docs.python.org/3/library/queue.html

v923z, 2020-10-28:
> I might be wrong here, but I think most user needs would be satisfied by implementing a queue function as exists in python https://docs.python.org/3/library/queue.html

First, that is a `python` function, and it has nothing to do with `numpy`, and `queue` has already been implemented in `micropython`.

But the more important question is, and I tried to articulate the difficulties above, what should happen with a circular buffer? How should array operators operate on them? To underline the seriousness of the problem, I am all but convinced that it is absolutely no accident that `numpy` has no circular buffer. You can go back to the original discussion in https://github.com/v923z/micropython-ulab/issues/14#issuecomment-577659945, and see why the problem is non-trivial. 

Finally, I would like to see a real-life example that requires a circular buffer. A different implementation for `roll` is not a pressing argument in my opinion.

v923z, 2020-10-30:
@roger- How do you want to resolve the issue? I think there are a couple of questions that should be clarified before we embark on implementing the circular buffer, but I don't know what your opinion is.

roger-, 2020-10-31:
@v923z 
One use case I have in mind is streaming sensor data into a buffer and processing it, where the buffers are too large to have multiple copies in memory. This would only be beneficial for reduction or in place operations though (`mean()`, `argmax()`, etc.), as most others return copies/views.

Don't think this can be done as a drop in replacement for `ndarray`, but it would be good to have a subset of the   `ndarray` methods supported. I'm thinking 1D, but perhaps it could be generalized.

A circular buffer `a` with max size 6 and 4 current items could look like this in memory:

```
X X X X X X
    ^     ^start
    |end
```

Then in ulab `a[n]` would map to `*(buffer + (start + n) % size)`, a[-1] would map to `*(buffer + end)`, etc. Think a good amount of numpy methods could apply, but I'm not familiar enough with ulab/numpy internals to say for sure. `append()`, `popleft()`, etc. methods might also be useful.

v923z, 2020-10-31:
> @v923z
> One use case I have in mind is streaming sensor data into a buffer and processing it, where the buffers are too large to have multiple copies in memory. This would only be beneficial for [...]

The main problem is that loops are linear. This doesn't matter for `mean`, `max`, `sum`, and the like, which do not care about the order of the data. But `argmax` is already problematic, and there are many more. Even more serious is the problem with `view`s: these are all based on the assumption that elements can be accessed via a linear combination of indices: https://micropython-ulab.readthedocs.io/en/latest/ulab-programming.html#memory-layout. The modulo operator upsets this concept. 

When you do anything with an array, you have a C loop that looks like (the `sum` function in this case)

```c
uint8_t *array = (uint8_t *)ndarray->array;
mp_float_t sum = 0.0;

for(size_t i=0; i < len; i++) {
    sum += *array;
    array += ndarray->strides[ULAB_MAX_DIMS-1];
}
```
We would have to split this into two loops (`sum` is not a particularly good example), like 

```c
uint8_t *array = (uint8_t *)ndarray->array;
array += ndarray->circular_pointer;

mp_float_t sum = 0.0;

for(size_t i=start; i < len; i++) {
    sum += *array;
    array += ndarray->strides[ULAB_MAX_DIMS-1];
}

uint8_t *array = (uint8_t *)ndarray->array;
for(size_t i=0; i < start; i++) {
    sum += *array;
    array += ndarray->strides[ULAB_MAX_DIMS-1];
}

```

> Don't think this can be done as a drop in replacement for `ndarray`, but it would be good to have a subset of the `ndarray` methods supported. I'm thinking 1D, but perhaps it could be generalized. [...]

In this whole problem indexing and slicing are the thorniest issues. 

> Think a good amount of numpy methods could apply, but I'm not familiar enough with ulab/numpy internals to say for sure. 

Could you compile a list of functions that would be interesting? Let's stay with the 1D case first!

> `append()`, `popleft()`, etc. methods might also be useful.

These actually modify the layout (they add or remove an element from the array), so these couldn't even be implemented as you suggest. To put it differently, with `append`, and `pop`, you can't avoid the `memcpy` operation that you wanted to circumvent in the first place. 

On the other hand, if you don't care about the actual position of the pointer, and you simply need a container that "behaves" like a circular buffer, you can already do that: the `.tobytes()` method gives you the raw data array that you can pass on to a DAC, or ADC or sound engine, or whatever, and you can do with that anything you want. https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html#tobytes So, if your DMA routine, which is independent of `ulab`, treats a string of raw bytes as a circular buffer, then you can fill up the array in the ADC, and process your data in `ulab`. 

To be a bit more constructive, here are some comments: the current implementation allows relatively flexible customisation, so I _could_, in principle, add code that implements a circular buffer, and you could switch that feature on in https://github.com/v923z/micropython-ulab/blob/master/code/ulab.h. We would then have to add an `insert` method, which would update your buffer, and then we would have to add code to some of the functions. I would definitely begin with the one dimension, and see how that works.

v923z, 2020-11-11:
@roger- Can you show an example of how you would use a circular buffer? I am trying to understand what you want to achieve.

street-grease-coder, 2020-11-11:
I think here you could find some examples and implementations that might to be relevant for makers https://github.com/andriyadi/MagicWand-TFLite-Arduino/tree/master/src

v923z, 2020-11-12:
I think what you linked to concerns the machine learning discussion https://github.com/v923z/micropython-ulab/issues/183. We should not mix subjects, because we will lose track of what is going on. 

The question is not really how to implement a circular buffer. That is almost trivial. But the buffer has to go into various functions, and there are no `numpy` examples as to what should happen then. 

And again, in order to see what is required here, we would need a conceptual skeleton, with functions that are needed. To be honest, there is very little in this discussion that would at least spell out the requirements.

roger-, 2020-11-12:
> These actually modify the layout (they add or remove an element from the array), so these couldn't even be implemented as you suggest. To put it differently, with append, and pop, you can't avoid the memcpy operation that you wanted to circumvent in the first place.

I envision them just moving points, so the size of the array could get larger or smaller (up to a max).

> So, if your DMA routine, which is independent of ulab, treats a string of raw bytes as a circular buffer

Yeah that's a good point. I guess ultimately if the underlying driver / hardware supports circular buffers then this would work with `tobytes()`. I was assuming that this was happening at the Python level though (which I guess is slow enough to make this all moot).

Here's a made up example:

```python
while True:
   val = sensor.read(1)
   buffer.insert(val)

   var = buffer.var()
   dft = np.fft.fft(buffer)
   # some other processing
```

Of course `fft`, `var`, etc. probably break with modulo indexing and would probably need to use some sort of additional abstraction.

v923z, 2020-11-12:
> I envision them just moving points, so the size of the array could get larger or smaller (up to a max).

But if you have to reserve new space each time you add a datum, then `roll` is not any worse. 

> Yeah that's a good point. I guess ultimately if the underlying driver / hardware supports circular buffers then this would work with `tobytes()`. I was assuming that this was happening at the Python [...]

Actually, `fft`, `var`, and the like are not too bad, because they  do not care about the order of items (as long as they come after each other). The phase of an `fft` will change, but if you care about the spectrum only, that doesn't matter. 

The problem is that there are some functions that are not so trivial: `argmax`, `polyfit`, `fft`, if you need the phase, any fitting routines, many of the linear algebra functions etc. It would lead to confusion, if we had an `ndarray` on steroids, and some functions couldn't take it as an argument. 

Then you have the problem of e.g., multiplying two such arrays, if they are not shifted by the same amount. I believe, this is a nasty can of worms. 

A possible implementation is under https://github.com/eric-wieser/numpy_ringbuffer/blob/master/numpy_ringbuffer/__init__.py . Note, however, that this is not efficient: you have to unwrap the buffer into a linear array, anytime you want to do some calculation. But if it is so, then it is better to take the `python` implementation.

roger-, 2020-11-12:
> But if you have to reserve new space each time you add a datum, then roll is not any worse.

I'm assuming there's a maximum size then gets allocated regardless. 

> Actually, fft, var, and the like are not too bad

Not the best example, but assume I'm doing windowing or something too. 

> I believe, this is a nasty can of worms.

Agreed, making an ndarray drop-in will be hard.

v923z, 2021-01-09:
Since there seems to be no activity on this issue, I am going to close it.

closed 2021-01-09 by v923z