v923z/micropython-ulab #187: Feature Request: adding equivalent to numpy.diag()

mouryarahul, 2020-10-28:
This function is useful in linear algebra, and I guess it won't take much of the flash memory. I think it should go to Linalg module.

v923z, 2020-10-28:
Do you mean this: https://numpy.org/doc/stable/reference/generated/numpy.diag.html ? 

You are right, it is relatively cheap, and doable. I can see to it, once 1.0 is out.

mouryarahul, 2020-10-28:
Yeah, Thanks!

v923z, 2020-10-29:
This would actually have to go into the `create` module, because it creates a new array. Also `linalg` is a sub-module, which you have to import even in `numpy`, and `diag` is not part of `linalg` in `numpy`.

mouryarahul, 2020-10-30:
I agree with you: it should be in the `create` as in Numpy.

v923z, 2020-10-31:
@mouryarahul I have just merged a commit resolving the issue. Could you, please, create a test script, and add that to https://github.com/v923z/micropython-ulab/tree/master/tests? Thanks!

mouryarahul, 2020-11-02:
`numpy.diag` works both ways: if it gets a 1D array, it returns a square matrix with those elements on its diagonal and if it receives a matrix, it returns a 1D array with the diagonal elements of the matrix. However, the current  `ulab.diagonal` implementation works in 2nd way only with some bugs. The bug is:

>>> x = np.array([[1,2,3],[4,5,6]]);
>>> x
array([[1.0, 2.0, 3.0],
       [4.0, 5.0, 6.0]], dtype=float)
>>> np.diagonal(x)
array([1.0, 5.0, 1.595373223098232e-154], dtype=float)

@v923z Sorry for the delay, I'm trying to write test scripts for all the modules. I hope to complete and submit it for review in few days.

v923z, 2020-11-02:
> `numpy.diag` works both ways: if it gets a 1D array, it returns a square matrix with those elements on its diagonal and if it receives a matrix, it returns a 1D array with the diagonal elements of the matrix. However, the current `ulab.diagonal` implementation works in 2nd way only

OK, this can be added, no problem. 

> The bug is: 

> > > > x = np.array([[1,2,3],[4,5,6]]);
> > > > x
> > > > array([[1.0, 2.0, 3.0],
> > > > [4.0, 5.0, 6.0]], dtype=float)
> > > > np.diagonal(x)
> > > > array([1.0, 5.0, 1.595373223098232e-154], dtype=float)

This is indeed an error. I re-open the issue.

v923z, 2020-11-02:
@mouryarahul I think the latest commit fixes the issue. Let me know, if you still find a glitch in it!

closed 2020-11-02 by v923z

mouryarahul, 2020-11-02:
@v923z Thanks, I did test it, so far no bug :+1: