v923z/micropython-ulab #188: add numpy.com

v923z, 2020-10-28:
https://numpy.org/doc/stable/reference/generated/numpy.cov.html

Starting point based on https://github.com/v923z/micropython-ulab/issues/183

v923z, 2021-02-17:
@street-grease-coder If you look at the notes section of https://numpy.org/doc/stable/reference/generated/numpy.cov.html, isn't everything you need already implemented in `ulab`?

street-grease-coder, 2021-02-17:
if there's a covariance function this should work indeed, sorry if I missed this! I will check it out but a bit short on time at the moment... sorry

v923z, 2021-02-17:
@street-grease-coder  What I meant is that you can calculate the covariance matrix in 5 `numpy` statements
 
```python
m = np.arange(10, dtype=np.float64)
>>> f = np.arange(10) * 2
>>> a = np.arange(10) ** 2.
>>> ddof = 1
>>> w = f * a
>>> v1 = np.sum(w)
>>> v2 = np.sum(w * a)
>>> m -= np.sum(m * w, axis=None, keepdims=True) / v1
>>> cov = np.dot(m * w, m.T) * v1 / (v1**2 - ddof * v2)
```

We might have to implement the `keepdims` keyword, though. Since the implementation of this function is in `python` even in `numpy`, I don't see, why we should move to C. 

Two similar issues have been raised in https://github.com/v923z/micropython-ulab/issues/320, and https://github.com/v923z/micropython-ulab/issues/278, and I think, we might have to add a `snippets` directory for `numpy` functions that are implemented in `python`. Do you want to try your hand at it?