v923z/micropython-ulab #191: Feature request: exposing the datatype of an array

mouryarahul, 2020-10-28:
We can already get the different attributes of an array, e.g., `shape`, `itemsize`, ... Will it be possible to expose somehow the **datatype** of an array element? Knowing the `itemsize` of an array, we cannot distinguish between datatype _int8_, or _uint8_ .

v923z, 2020-10-29:
We could do that, though it is a bit trickier than you would think. I assume you want to "copy" the `dtype` like so

```
a = array([1, 2, 3])
b = array([4, 5, 6], dtype=a.dtype)
```

mouryarahul, 2020-10-30:
Yeah, I was temporarily (until we will have in ULAB) implementing `numpy.diag` and needed to create an output array with the same datatype.

v923z, 2020-10-30:
> Yeah, I was temporarily (until we will have in ULAB) implementing `numpy.diag` 

If you can post the code, we could move faster...

mouryarahul, 2020-10-30:
What I mean was that I was implementing it in Pythonic way, which is a bit crude not with all the options as in the Numpy:
Sorry, some part of the code are unformatted.

def diag(a: ulab.array) -> ulab.array:
    assert isinstance(a, ulab.array), "Input should be of type ulab.array!"
    m, n = a.shape()
    itemsize = a.itemsize()
    if itemsize == 1:
        dtype = ulab.int8
    elif itemsize == 2:
        dtype = ulab.int16
    elif itemsize == 8:
        dtype = ulab.float
    else:
        raise Exception("Unknown datatype of the input array!")

    if m <= 1 and n <= 1:
        return a
    elif (m == 1 and n > 1) or (m > 1 and n == 1):
        r = m if m > n else n
        b = ulab.eye(r, dtype=dtype)
        for i in range(r):
            b[i, i] = a[i]
        return b
    elif m == n:
        b = ulab.zeros(m, dtype=dtype)
        for i in range(m):
            b[i] = a[i][i]
        return b
    else:
        raise Exception("Input should be square matrix!")

v923z, 2020-10-30:
I think I can release 1.0 now, so I should be able to implement a function for getting the `dtype` by the beginning of next week at the latest.

mouryarahul, 2020-10-31:
Magnificent work! Congratulation...

v923z, 2020-11-03:
I have a working implementation of `dtype` now, but I still need a bit of time to figure out, how it could be incorporated into `circuitpython`. They have a notion of properties, while `micropython` doesn't, hence the holdup.

v923z, 2020-11-10:
@mouryarahul See the comments https://github.com/v923z/micropython-ulab/issues/208#issuecomment-724255578 and thereafter.

v923z, 2020-11-11:
https://github.com/v923z/micropython-ulab/pull/216 implements the requested feature, so I am closing now.

closed 2020-11-11 by v923z