v923z/micropython-ulab #192: Possible glitch in indexing and modifying the element of an array

mouryarahul, 2020-10-28:
Is this intentional or some limitation?
In  
`>>> ulab.__version__`
`'0.54.5'`

While accessing an element:
`>>> import ulab as np`
`>>> X = np.array([[1,2,3],[4,5,6],[7,8,9]])`
`>>> X[0][0]`
`1.0`
`>>> X[0,0]`
`array([1.0], dtype=float)`

While modifying the element:

`>>> X[0][0] = 10`
`>>> X`
`array([[1.0, 2.0, 3.0],`
`         [4.0, 5.0, 6.0],`
`         [7.0, 8.0, 9.0]], dtype=float)`
`>>> X[0,0] = 10.`
`>>> X`
`array([[10.0, 2.0, 3.0],`
`         [4.0, 5.0, 6.0],`
`         [7.0, 8.0, 9.0]], dtype=float)`

While on CPython (NumPy), the two indexing styles have the same output and the element can be modified in both indexing ways.

v923z, 2020-10-29:
In version 0.54, `ulab` simply creates a copy, because it has no notion of views. That is what version 1.0 is about.

v923z, 2020-10-30:
Version 1.0.0 is out, so the issue should be resolved.

mouryarahul, 2020-11-09:
I guess the issue is still not solved yet.
Indexing a single element of an array returns an array of a single element. See below:

`>>> a`
`array([[1.0, 2.0, 10.0],
       [4.0, 5.0, 6.0],
       [7.0, 8.0, 9.0]], dtype=float)`
`>>> a[0,0]`
`array([[1.0]], dtype=float)`
`>>> a[0][0]`
`array([[1.0, 2.0, 10.0]], dtype=float)`

What will be a possible way to get a single element of an array?

mouryarahul, 2020-11-09:
Moreover, I find a small glitch in slicing an array:

>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]])
>>> a[3:3,:]
array([0.0, 0.0, 0.0]], dtype=float)
>>> a[:,3:3]
array([[],
       [],
       []], dtype=float)

While in `Numpy`, it returns with an empty array single dimension:
>>> a[:,3:3]
array([], shape=(3, 0), dtype=int64)
>>> a[3:3,:]
array([], shape=(0, 3), dtype=int64)

So, I think the glitch is in the first case. 

I'm still writing/testing other cases and other modules.

mouryarahul, 2020-11-09:
Moreover, I find a small glitch in slicing an array:

>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]])
>>> a[3:3,:]
array([0.0, 0.0, 0.0]], dtype=float)
>>> a[:,3:3]
array([[],
       [],
       []], dtype=float)

While in `Numpy`, it returns with an empty array single dimension:
>>> a[:,3:3]
array([], shape=(3, 0), dtype=int64)
>>> a[3:3,:]
array([], shape=(0, 3), dtype=int64)

So, I think the glitch is in the first case.

v923z, 2020-11-09:
> I guess the issue is still not solved yet.
> Indexing a single element of an array returns an array of a single element. See below:
> 
> `>>> a`
> `array([[1.0, 2.0, 10.0], [4.0, 5.0, 6.0], [7.0, 8.0, 9.0]], dtype=float)`
> `>>> a[0,0]`
> `array([[1.0]], dtype=float)`
> `>>> a[0][0]`
> `array([[1.0, 2.0, 10.0]], dtype=float)`
> 
> What will be a possible way to get a single element of an array? In `Numpy` both type of indexing is supported. I think we can keep `a[i, j]` style.

When you say `a[3][2][1][0]`, you have only a single index, 0. The rest are producing views of your array, and as such, you can't just say that you want to have only `a[i,j]`, and forgo `a[i][j]`. Both should work.

mouryarahul, 2020-11-09:
> When you say `a[3][2][1][0]`, you have only a single index, 0. The rest are producing views of your array, and as such, you can't just say that you want to have only `a[i,j]`, and forgo `a[i][j]`. [...]

Well, if we follow `Numpy` style, then both should work, otherwise, in languages e.g., Matlab, there is only one way:
`a(i,j,k,...)`. But, let stick to `Numpy` style.

v923z, 2020-11-10:
> Well, if we follow `Numpy` style, then both should work, otherwise, in languages e.g., Matlab, there is only one way:
a(i,j,k,...). 

`matlab` should have absolutely no relevance in this case. But I hope this https://github.com/v923z/micropython-ulab/pull/211 fixes it.

closed 2020-11-10 by v923z