v923z/micropython-ulab #193: Docu request

street-grease-coder, 2020-10-30:
Hi... sorry for the very silly question, but somehow I seem to not be able to just 'import ulab'. Is it possible that ulab is not contained in the standard bundle of micropython at install? I didn't find this here in the help or on micropython docu... sorry again for the extremely obvious question

v923z, 2020-10-30:
`ulab` is not part of `micropython` itself, and that has many reasons. But you still have a couple of options. You can try Roberto Colistete's repo, where he uploads compiled firmware for a large number of platforms on a regular bases. Or you can try to compile it yourself. Here are the instructions: https://github.com/v923z/micropython-ulab/tree/master#firmware

street-grease-coder, 2020-10-30:
this is more complicated than anticipated, wow. I guess there's no way to just upip install ulab onto my ESP32. I guess there's no way around compiling the whole thing yourself if you want advanced features. I don't know what the differences are from 1.12 and 1.13 micropython, but Roberto's Gitlab page is still on 1.12. I'll try that first... then, under windows, I fully anticipate the compilation of micropython (and adding ulab) will probably be a frustrating procedure. Thanks for the quick reply for now!

v923z, 2020-10-30:
> this is more complicated than anticipated, wow. I guess there's no way to just upip install ulab onto my ESP32. I guess there's no way around compiling the whole thing yourself if you want advanced features. 

The issue is really complicated: `ulab` adds to the firmware. Some platforms don't have enough space, so you would have to compromise, what you are willing to sacrifice. You want to have `ulab`, or `threading` support etc. This is partly what Roberto is trying to do, but it is a very difficult problem. 

Second, Damien George wants to have full control of the `micropython` firmware in the sense that he wants to be able to guarantee the functionality and quality of it. Once you start to incorporate third-party libraries, like `ulab`, you loose this kind of control. 

>I don't know what the differences are from 1.12 and 1.13 micropython, but Roberto's Gitlab page is still on 1.12. 

Here are the change logs: https://github.com/micropython/micropython/releases

street-grease-coder, 2020-10-30:
thanks for the kind and patient explanations, I hope I can contribute to this project someday, in some way to return the favor.

closed 2020-10-30 by street-grease-coder