v923z/micropython-ulab #195: Compilation error when #define ULAB_MAX_DIMS     4

mouryarahul, 2020-11-01:
I got compilation error when `#define ULAB_MAX_DIMS                   4`, while it compiles without error when `#define ULAB_MAX_DIMS                   3`.
The error msg is the following:

_make BOARD=PYBD_SF6 USER_C_MODULES=../../../micropython-ulab all
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../../micropython-ulab/code
CC ../../../micropython-ulab/code/numerical/numerical.c
../../../micropython-ulab/code/numerical/numerical.c: In function 'numerical_argmin_argmax_ndarray':
../../../micropython-ulab/code/numerical/numerical.c:260:13: error: 'sarray' undeclared (first use in this function); did you mean 'array'?
  260 |             sarray -= ndarray->strides[ULAB_MAX_DIMS - 3] * ndarray->shape[ULAB_MAX_DIMS-3];
      |             ^~~~~~
      |             array
../../../micropython-ulab/code/numerical/numerical.c:260:13: note: each undeclared identifier is reported only once for each function it appears in
make: *** [../../py/mkrules.mk:63: build-PYBD_SF6/code/numerical/numerical.o] Error 1_

v923z, 2020-11-01:
This was a copy-paste error in the last line of the loop. Thanks!

mouryarahul, 2020-11-01:
Now, it does compile successfully, but many of the existing tests fail in the case when `#define ULAB_MAX_DIMS  3 or 4`. 
The test msg:

FAIL  tests/00smoke.py
pass  tests/argminmax.py
FAIL  tests/cholesky.py
pass  tests/compare.py
FAIL  tests/constructors.py
FAIL  tests/linalg.py
pass  tests/operators.py
pass  tests/poly.py
pass  tests/slicing.py
pass  tests/slicing2.py
10 tests performed (1084 individual testcases)
6 tests passed
4 tests failed: 00smoke cholesky constructors linalg

I'm already writing extensive "Test Scripts", which I'll submit to the repo in the next few days for the review.

v923z, 2020-11-02:
@mouryarahul  We should re-open the issue, this is serious.

v923z, 2020-11-02:
@mouryarahul  I believe, this is fixed in the latest, so I am closing the issue for now. Open a new ticket, if this is still wrong.

closed 2020-11-02 by v923z