v923z/micropython-ulab #200: Bug in Array Creation Module

mouryarahul, 2020-11-03:
There is an error in array creation module, I guess, which is also reflected in Linalg module too. See below:

`MicroPython v1.13 on 2020-11-02; linux version`
`Use Ctrl-D to exit, Ctrl-E for paste mode`
`>>> import ulab as np`
`>>> np.__version__`
`'1.1.3-2D-numpy'`
`>>> x = np.array([1,2,3])`
`>>> x.shape()`
`(0,)`
`>>> np.linalg.norm(x)`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`ValueError: norm is defined for 1D and 2D arrays`
`>>> `
`>>> y = np.array([[1,2,3]])`
`>>> y.shape()`
`(1, 3)`
`>>> np.linalg.norm(y)`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`ValueError: norm is defined for 1D and 2D arrays`
`>>> `

v923z, 2020-11-03:
@mouryarahul Oh, bollocks, these are actually trivial errors, (It is still bad, that they are there.) so if you want to, you can fix them yourself. I won't have time till the evening. 

`norm` fails because of a false condition. In https://github.com/v923z/micropython-ulab/blob/41320fc1b3a03373c3f05c8d6a2475af68bbdf17/code/linalg/linalg.c#L491 you should have 
```c
if((ndarray->ndim != 1) && (ndarray->ndim != 2))
```

As for the `shape`, the problem there is that the shape values are listed backwards. In the `ndarray_shape` function at https://github.com/v923z/micropython-ulab/blob/41320fc1b3a03373c3f05c8d6a2475af68bbdf17/code/ndarray.c#L1409 you have to insert the values like so 
```c
 for(uint8_t i=0; i < self->ndim; i++) {
        items[self->ndim - i - 1] = mp_obj_new_int(self->shape[ULAB_MAX_DIM - i - 1]);
    }
```
This is detailed in https://micropython-ulab.readthedocs.io/en/latest/ulab-programming.html#iterating-over-elements-of-a-tensor, and you find a couple of examples in https://micropython-ulab.readthedocs.io/en/latest/ulab-programming.html#creating-a-new-ndarray. 
Sorry that I can't be of more help at the moment.

mouryarahul, 2020-11-03:
Thanks! I'll try fixing it.

mouryarahul, 2020-11-03:
I implemented your suggestions. This seems to fix the `shape` bug, but the `linalg.norm` is not fixed yet. I get following:
>>> import ulab as np
>>> x = np.array([1,2,3])
>>> x.shape()
(3,)
>>> np.linalg.norm(x)
Segmentation fault (core dumped)

I do not want to dig into the code, so I wait until you fix it.

v923z, 2020-11-03:
Well, this is quite embarrassing, sorry for the trouble! In https://github.com/v923z/micropython-ulab/blob/41320fc1b3a03373c3f05c8d6a2475af68bbdf17/code/linalg/linalg.c#L504-L505 the two indices are swapped. One should have 

```c
        array -= ndarray->strides[ULAB_MAX_DIMS - 1] + ndarray->shape[ULAB_MAX_DIMS - 1];
        array += ndarray->strides[ULAB_MAX_DIMS - 2];
```
instead. But if you don't feel like fixing it, I can take care of it later today.

mouryarahul, 2020-11-03:
I will try your suggestion again, let you know. Thanks!

closed 2020-11-03 by v923z