v923z/micropython-ulab #205: Small glitch in ulab.sum and ulab.mean

mouryarahul, 2020-11-04:
Is this intentional?

MicroPython v1.13 on 2020-11-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab as np
>>> a = np.array([[1, 12, 3, 0], [5, 3, 4, 1], [9, 11, 1, 8], [7, 10, 0, 1]], dtype=np.float)
>>> np.sum(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: operation is not implemented for flattened array
>>> np.mean(a)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: operation is not implemented for flattened array
>>> np.median(a)
3.5
>>>

v923z, 2020-11-04:
Well, if it is not implemented, it can't be erroneous, can it?

The default value of `axis` is `None`, and that corresponds to the flattened array. Indeed, that case is not implemented. The problem is, it is not difficult, but not very cheap (in code size), either, and there are at least two and a half ways of achieving what you want. You can flatten your array, before you pass it to `sum` (this will produce a new copy), you can calculate the sum in two passes 

```python
sum(sum(a, axis=1), axis=0)
```
(that will produce a smaller, contracted array), and the half way is to reshape your array first, as in `a.reshape((1,12))`. That works, if the array is dense, is fast, and is economical with RAM, because re-shaping doesn't produce a copy, only a view. (For dense arrays, that is.)

So, not implementing the flat case was some sort of trade-off. I could add it, if you think it is important, but I don't perceive this to be a pressing issue. I could implement it via function calls (instead of resolving all 5 `dtype`s. That doesn't add too much to the firmware (probably less than a kB for all three of `sum`, `mean`, and `std` combined), but is somewhat slower than a full-blown implementation. Would that be acceptable?

One final comment: I have discussed functions/functionality in `ulab` with quite a few people, and the consensus seems to be that much more is implemented than what is actually used. (Most were very excited to see a library that can do FFT in `micropython`, but don't give a hoot about the Cholesky decomposition, e.g.) I am more than happy to add new functions, if they are used, even if only by a handful of people in the whole world. But if the only argument in favour of something new is that it makes `ulab` more `numpy`-like, then we have to think twice. My resources, and that of a microcontroller are limited. Incidentally, these two statements are linked: much effort goes into implementing a seemingly trivial routine in a way that doesn't use up all the flash, and doesn't stress the RAM.

I would really like to see a somewhat more organised way of polling user opinion. If you look back at past feature requests, most of them were raised by a single person, and no-one even commented, so it is a bit hard for me to judge what is important/useful/beneficial. If you have some ideas as to how testing user experience could be carried out, I am all ears.

mouryarahul, 2020-11-04:
First, let mention that what you have been doing for the Micropython Community is really magnificent work. Almost all the functions implemented in ULAB are especially useful for anyone building applications based on numerical computing, e.g., in my case Signal processing applications. In my opinion, basic linear algebra, statistics, and some signal processing like FFT, are sufficient for ULAB. Anyone who wishes to build some complex algorithm can happily use these basic functions. I agree with you that we cannot make ULAB like NumPy (adding many functions from it), but I will be in favour that whatever functions are inspired from NumPy, should have similar input/output behaviour and most of its functions are already compatible. Now, with the release of v1.x.x, ULAB is a full-blown numerical library for Micropython with great online documentation :+1:. I guess in a few weeks, we can eliminate most of the glitches too. Though I do not contribute to writing the library itself, I hope my feedbacks are helpful in improving it.

Btw, Micropython's `sum` function does the job, but still, I find a glitch in `ulab.sum` and `ulab.mean`:
MicroPython v1.13 on 2020-11-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ulab as np
>>> np.__version__
'1.2.1-2D-numpy'
>>> a = np.array([[1, 12, 3, 0], [5, 3, 4, 1], [9, 11, 1, 8], [7, 10, 0, 1]], dtype=np.float)
>>> sum(a)
array([22.0, 36.0, 8.0, 10.0], dtype=float)
>>> sum(sum(a))
76.00000000000001
>>> np.sum(np.sum(a, axis=1), axis=0)
Segmentation fault (core dumped)

>>> np.mean(a, axis=0)
array([5.5, 9.0, 2.0, 2.5], dtype=float)
>>> np.mean(a, axis=1)
array([4.0, 3.25, 7.25, 4.5], dtype=float)
>>> np.mean(np.mean(a, axis=1), axis=0)
Segmentation fault (core dumped)

The `ulab.median` function works well in all cases:

>>> np.median(a)
3.5
>>> np.median(a, axis=1)
array([2.0, 3.5, 8.5, 4.0], dtype=float)
>>> np.median(a, axis=0)
array([6.0, 10.5, 2.0, 1.0], dtype=float)
>>> np.median(np.median(a, axis=1), axis=0)
3.75

v923z, 2020-11-04:
> I guess in a few weeks, we can eliminate most of the glitches too. Though I do not contribute to writing the library itself, I hope my feedbacks are helpful in improving it.

Definitely, and I really appreciate your comments. The thing is, I can't possibly foresee all use cases, and that means that I can't possibly test for all kinds of scenarios. In fact, I think one should never test alone. 

> Btw, Micropython's `sum` function does the job, but still, I find a glitch in `ulab.sum` and `ulab.mean`:
> MicroPython v1.13 on 2020-11-04; linux version
> Use Ctrl-D to exit, Ctrl-E for paste mode

I will look into that. I can also add the case for the flat array. It is really not that difficult.

mouryarahul, 2020-11-05:
I think, `ulab.sum` without any further argument should be the default method. So, it will be good to add it.

v923z, 2020-11-06:
@mouryarahul I think I have fixed the issue.

mouryarahul, 2020-11-08:
Now, `ulab.sum`, `ulab.mean` and `ulab.median` seem to work well. So, I think we can close the issue.

v923z, 2020-11-08:
We should keep this open, because the flat array is still not treated in `sum`.

v923z, 2020-11-09:
https://github.com/v923z/micropython-ulab/pull/209 implements the treatment of `axis=None`, so closing now.

closed 2020-11-09 by v923z