v923z/micropython-ulab #208: Array creation glitch: while creating an array from another array

mouryarahul, 2020-11-09:
I'm still testing and writing test scripts. Meanwhile, I find some glitch in array creation module. I guess it is something to do with the casting. See below for the probable glitch:

`>>>MicroPython v1.13 on 2020-11-08; linux version`
`Use Ctrl-D to exit, Ctrl-E for paste mode`
`>>> import ulab as np`
`>>> np.__version__`
`'1.2.3-2D-numpy'`
`>>> print(np.array(np.array(range(5), dtype=np.uint8), dtype=np.uint8))`
`array([0, 1, 2, 3, 4], dtype=uint8)`
`>>> print(np.array(np.array(range(5), dtype=np.uint8), dtype=np.int8))`
`array([0, 0, 0, 0, 0], dtype=int8)`
`>>> print(np.array(np.array(range(5), dtype=np.uint8), dtype=np.uint16))`
`array([0, 0, 0, 0, 0], dtype=uint16)`
`>>> print(np.array(np.array(range(5), dtype=np.uint8), dtype=np.int16))`
`array([0, 0, 0, 0, 0], dtype=int16)`
`>>> print(np.array(np.array(range(5), dtype=np.uint8), dtype=np.float))`
`array([0.0, 0.0, 0.0, 0.0, 0.0], dtype=float)`
`>>> print(np.array(np.array(range(5), dtype=np.int8), dtype=np.uint8))`
`array([0, 0, 0, 0, 0], dtype=uint8)`
`>>> print(np.array(np.array(range(5), dtype=np.int8), dtype=np.int8))`
`array([0, 1, 2, 3, 4], dtype=int8)`
`>>> print(np.array(np.array(range(5), dtype=np.int8), dtype=np.uint16))`
`array([0, 0, 0, 0, 0], dtype=uint16)`
`>>> print(np.array(np.array(range(5), dtype=np.int8), dtype=np.int16))`
`array([0, 0, 0, 0, 0], dtype=int16)`
`>>> print(np.array(np.array(range(5), dtype=np.int8), dtype=np.float))`
`array([0.0, 0.0, 0.0, 0.0, 0.0], dtype=float)`

and other combinations also follow the above trend.

In `Numpy`, all these produces intended results.

Moreover, I found this small glitch:
`>>> np.float`
`100`

v923z, 2020-11-09:
> I'm still testing and writing test scripts. Meanwhile, I find some glitch in array creation module. I guess it is something to do with the casting. See below for the probable glitch: [...]

Right. Thanks for reporting the issue! This should be fixed in https://github.com/v923z/micropython-ulab/pull/210

> `>>> np.float`
> `100`

100 is the type code for float (ASCII `d`, which is `double` in `python`), so this is not a glitch, this is a cheap way of having `dtype`. Although `dtype` has been implemented for `micropython`, I would probably still keep this as an acceptable workaround (you notice the difference to `numpy` only, if you print out the value of `np.float`, otherwise, it does what it is supposed to), because the proper implementation of `dtype` adds close to a kB to the firmware, so that might not be a viable option on smaller MCUs. 

If `np.float = 100` disturbs you, and you are willing to pay a kB for a fix, you will be able to set a switch in `ulab.h`. 

> `>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]], dtype=np.int8)`
> `>>> float(a)`
> `Segmentation fault (core dumped)`
> 
> while this is OK:
> 
> `>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]], dtype=np.int8)`
> `>>> int(a)`
> `Traceback (most recent call last):`
> ` File "<stdin>", line 1, in <module>`
> `TypeError: can't convert ndarray to int`
> 
> I was wondering how to test any possible memory leakage problem?

This should probably be raised at `micropython`. I think I know what the problem is, and it really has to be fixed in `micropython` itself.

closed 2020-11-09 by v923z

v923z, 2020-11-09:
@jepler Jeff, I have opened a ticket at `micropython`, but I think `circuitpython` is the same, 
```
>>> a = np.array([[1,2,3],[4,5,6],[7,8,9]], dtype=np.int8)
>>> float(a)
```
will produce a seg fault.

mouryarahul, 2020-11-09:
> Right. Thanks for reporting the issue! This should be fixed in #210
>
>
> 100 is the type code for float (ASCII `d`, which is `double` in `python`), so this is not a glitch, this is a cheap way of [...]

I'm fine with it, but do we need to expose it? In `Numpy`, `np.int8`, `np.int16`, `np.float32` are also implemented as callable functions, which convert a scalar to its type and convert a numeric list to an array of its datatype. 

> This should probably be raised at `micropython`. I think I know what the problem is, and it really has to be fixed in `micropython` itself.

v923z, 2020-11-10:
> I'm fine with it, but do we need to expose it? In `Numpy`, `np.int8`, `np.int16`, `np.float32` are also implemented as callable functions, which convert a scalar to its type and convert a numeric [...]

Is it correct, that these cases are really nothing but another way of writing 

```
np.float32([1, 2, 3, 4]) <=> ulab.array([1, 2, 3, 4], dtype=ulab.float)
```
If you want, I can add a wrapper for these, and then we could simply call the creation array operator. That would be cheap, because we pay for the `QSTR`.

mouryarahul, 2020-11-10:
> > > > `>>> np.float`
> > > > `100`
> > > 
> > > 
> > > 100 is the type code for float (ASCII `d`, which is `double` in `python`), so this is not a glitch, this is a cheap way of having `dtype`. Although `dtype` has been implemented for `micropython`, I would probably still keep this as an acceptable workaround (you notice the difference to `numpy` only, if you print out the value of `np.float`, otherwise, it does what it is supposed to), because the proper implementation of `dtype` adds close to a kB to the firmware, so that might not be a viable option on smaller MCUs.
> > > If `np.float = 100` disturbs you, and you are willing to pay a kB for a fix, you will be able to set a switch in `ulab.h`.
> > 
> > 
> > I'm fine with it, but do we need to expose it? In `Numpy`, `np.int8`, `np.int16`, `np.float32` are also implemented as callable functions, which convert a scalar to its type and convert a numeric list to an array of its datatype.
> 
> Is it correct, that these cases are really nothing but another way of writing
> 
> ```
> np.float32([1, 2, 3, 4]) <=> ulab.array([1, 2, 3, 4], dtype=ulab.float)
> ```
> 
> If you want, I can add a wrapper for these, and then we could simply call the creation array operator. That would be cheap, because we pay for the `QSTR` only. But we should make this configurable.

I think it is not necessary.