v923z/micropython-ulab #218: List of probable minor glitches

mouryarahul, 2020-11-15:
Below is the list of some minor glitches present in recent version. Though I've not printed the outputs, but you can easily figure out the error if compare the results with Numpy.

# List of Glitches

1.
a = np.array([1,2,3], dtype=np.float)
a **= np.array([4,5,6], dtype=np.uint16)
print(a)

2.
a = np.array([1,2,3], dtype=np.float)
a /= np.array([4,5,6], dtype=np.uint16)
println(a)

3. Minor glitch: returns one element array
println(np.clip(5, 3, 6))

4. Glitch in np.roll
println(np.arange(10))
println(np.roll(np.arange(10), 2))
println(np.roll(np.arange(10), -2))
a = np.array([1, 2, 3, 4, 5, 6, 7, 8])
println(np.roll(a, 2))
println(np.roll(a, -2))

5. Probable error: outputs are not equal to Numpy
a = np.array([253, 254, 255], dtype=np.uint8)
println(np.std(a))
println(np.std(a, axis=0))  
a = np.array([range(255-3, 255),range(240-3, 240),range(250-3,250)], dtype=np.float)
println(np.std(a))
println(np.std(a, axis=0))
println(np.std(a, axis=1))

6. Minor floating-point error
a = np.array([253, 254, 255], dtype=np.uint8)
println(np.sum(a))
println(np.sum(a, axis=0))
a = np.array([range(255-3, 255),range(240-3, 240),range(250-3,250)], dtype=np.float)
println(np.sum(a))
println(np.sum(a, axis=0))  ## Here is problem: minor floating point error
println(np.sum(a, axis=1))  ## Here is problem: minor floating point error

7. Minor glitch in flatten array case
a = np.array([range(255-5, 255),range(240-5, 240),range(250-5,250)], dtype=np.int8)
println(np.max(a)) ## Problem here

8. Minor glitch in flatten array case
a = np.array([range(255-5, 255),range(240-5, 240),range(250-5,250)], dtype=np.int8)
println(np.min(a)) ## Problem here

9. Glitch in np.trace
print(np.linalg.trace(np.eye(3)))

10. Glitch in case of non-flat array input
a = np.array([[1, 2, 3], [4, 5, 6], [7, 8, 9]])
print(np.linalg.norm(a)) 

11.
b = np.array([[1, 2], [3, 4]], dtype=np.int8)
print(np.linalg.size(b, axis=None))
print(np.linalg.size(b, axis=0))
print(np.linalg.size(b, axis=1))

12. Glitch in np.convolve
x = np.array((1,2,3))
y = np.array((1,10,100,1000))
print(np.convolve(x, y))

13. Probable error in np.sosfilt. I'm not sure yet.

14. Glitch in np.interp: output do not match to Numpy
x = np.array([1, 2, 3, 4, 5])
xp = np.array([1, 2, 3, 4])
fp = np.array([1, 2, 3, 4])    
print(np.interp(x, xp, fp))  ## Here is problem
print(np.interp(x, xp, fp, left=0.0))
print(np.interp(x, xp, fp, right=10.0))
print(np.interp(x, xp, fp, left=0.0, right=10.0))

15. Glitch in np.trapz: output do not match to Numpy
x = np.linspace(0, 9, num=10)
y = x*x
print(np.trapz(y))
print(np.trapz(y, x=x))

16. Not implemented but present in the documention
x = np.array([1,2,3])
x.sort()

mouryarahul, 2020-11-15:
I have almost completed the scripts for testing ULAB. I'll do a pull request soon.

kbrenner21, 2020-11-16:
@mouryarahul , have you noticed any issues with just trying to splice/index an np array? I am using the PYBD-SF6W and have noticed that I cannot access the actual integer/float entries within a given np array. Here is an example:

>>> import ulab as np
>>> a = np.zeros((3,3))
>>> a
array([[0.0, 0.0, 0.0],
[0.0, 0.0, 0.0],
[0.0, 0.0, 0.0]], dtype=float)
>>> a[0]
array([[0.0, 0.0, 0.0]], dtype=float)
>>> a[0][0]
array([[0.0, 0.0, 0.0]], dtype=float)

mouryarahul, 2020-11-16:
@kbrenner21 I guess you are using a bit older version of the firmware. This problem is rectified in the current version:
`>>> np.__version__`
`'1.3.1-2D-numpy'`
Here is the output on current version:
`>>> a = np.zeros((3,3))`
`>>> a`
`array([[0.0, 0.0, 0.0],
       [0.0, 0.0, 0.0],
       [0.0, 0.0, 0.0]], dtype=float64)`
`>>> a[0]`
`array([0.0, 0.0, 0.0], dtype=float64)`
`>>> a[0][0]`
`0.0`

v923z, 2020-11-16:
@mouryarahul Thanks for the extensive list. I will try to fix these in the coming days. 

As for the test scripts, it might make sense to commit and PR them one by one, but since I don't see how you organised them, it might not be the case.

v923z, 2020-11-16:
@mouryarahul 

> 3. Minor glitch: returns one element array
> print(np.clip(5, 3, 6))

I have just checked, this indeed returns `array([5], dtype=uint8)`. However, `numpy`'s help is not very clear: https://numpy.org/doc/stable/reference/generated/numpy.clip.html

In any case, I will fix it, so that it conforms to `numpy`'s code, and not the documentation.

v923z, 2020-11-17:
@mouryarahul I have added checkboxes to your list, so that we can keep track of what is going on. 

# List of Glitches

- [x] 1.
a = np.array([1,2,3], dtype=np.float)
a **= np.array([4,5,6], dtype=np.uint16)
print(a)

Fixed in https://github.com/v923z/micropython-ulab/pull/224

- [x] 2.
a = np.array([1,2,3], dtype=np.float)
a /= np.array([4,5,6], dtype=np.uint16)
println(a)

Fixed in https://github.com/v923z/micropython-ulab/pull/224

- [x] 3. Minor glitch: returns one element array
println(np.clip(5, 3, 6)) 

Fixed in https://github.com/v923z/micropython-ulab/pull/221

- [x] 4. Glitch in np.roll
println(np.arange(10))
println(np.roll(np.arange(10), 2))
println(np.roll(np.arange(10), -2))
a = np.array([1, 2, 3, 4, 5, 6, 7, 8])
println(np.roll(a, 2))
println(np.roll(a, -2))

Fixed in https://github.com/v923z/micropython-ulab/pull/224
- [ ] 5. Probable error: outputs are not equal to Numpy
a = np.array([253, 254, 255], dtype=np.uint8)
println(np.std(a))
println(np.std(a, axis=0))  
a = np.array([range(255-3, 255),range(240-3, 240),range(250-3,250)], dtype=np.float)
println(np.std(a))
println(np.std(a, axis=0))
println(np.std(a, axis=1))

- [ ] 6. Minor floating-point error
a = np.array([253, 254, 255], dtype=np.uint8)
println(np.sum(a))
println(np.sum(a, axis=0))
a = np.array([range(255-3, 255),range(240-3, 240),range(250-3,250)], dtype=np.float)
println(np.sum(a))
println(np.sum(a, axis=0))  ## Here is problem: minor floating point error
println(np.sum(a, axis=1))  ## Here is problem: minor floating point error

- [ ] 7. Minor glitch in flatten array case
a = np.array([range(255-5, 255),range(240-5, 240),range(250-5,250)], dtype=np.int8)
println(np.max(a)) ## Problem here

- [ ] 8. Minor glitch in flatten array case
a = np.array([range(255-5, 255),range(240-5, 240),range(250-5,250)], dtype=np.int8)
println(np.min(a)) ## Problem here

- [x] 9. Glitch in np.trace
print(np.linalg.trace(np.eye(3))) 

Fixed in https://github.com/v923z/micropython-ulab/pull/222

- [ ] 10. Glitch in case of non-flat array input
a = np.array([[1, 2, 3], [4, 5, 6], [7, 8, 9]])
print(np.linalg.norm(a)) 

- [ ] 11.
b = np.array([[1, 2], [3, 4]], dtype=np.int8)
print(np.linalg.size(b, axis=None))
print(np.linalg.size(b, axis=0))
print(np.linalg.size(b, axis=1))

- [ ] 12. Glitch in np.convolve
x = np.array((1,2,3))
y = np.array((1,10,100,1000))
print(np.convolve(x, y))

- [ ] 13. Probable error in np.sosfilt. I'm not sure yet.

- [ ] 14. Glitch in np.interp: output do not match to Numpy
x = np.array([1, 2, 3, 4, 5])
xp = np.array([1, 2, 3, 4])
fp = np.array([1, 2, 3, 4])    
print(np.interp(x, xp, fp))  ## Here is problem
print(np.interp(x, xp, fp, left=0.0))
print(np.interp(x, xp, fp, right=10.0))
print(np.interp(x, xp, fp, left=0.0, right=10.0))

- [ ] 15. Glitch in np.trapz: output do not match to Numpy
x = np.linspace(0, 9, num=10)
y = x*x
print(np.trapz(y))
print(np.trapz(y, x=x))

- [ ] 16. Not implemented but present in the documention
x = np.array([1,2,3])
x.sort()

- [x] 17. In `np.eye(N, M, k=0, dtype=float)`, any particular reason for M as rows and N as columns, which is just opposite of Numpy? 

Fixed in https://github.com/v923z/micropython-ulab/pull/223

v923z, 2020-11-17:
@mouryarahul As for issue 6, your problem is that 

```python
import ulab as np

print(np.__version__)

a = np.array([1,2,3], dtype=np.float)
b = np.array([4,5,6], dtype=np.uint16)

a = np.array([range(255-3, 255),range(240-3, 240),range(250-3,250)], dtype=np.float)
print(a)
print(np.sum(a))
print(np.sum(a, axis=0)) ## Here is problem: minor floating point error
print(np.sum(a, axis=1)) ## Here is problem: minor floating point error
```

results in 

```python
1.3.7-2D-numpy
array([[252.0, 253.0, 254.0],
       [237.0, 238.0, 239.0],
       [247.0, 248.0, 249.0]], dtype=float64)
2217.0
array([736.0, 739.0000000000001, 742.0], dtype=float64)
array([759.0, 714.0000000000001, 744.0], dtype=float64)
```
and you don't like the long 714.0000000000001 and similar numbers, right? I believe, this is a problem with the resolution of floating point numbers, and I am not sure there is too much we can do about it. In any case, the printout comes from `micropython`, so we might have to raise the issue there. You can see this here: 

```python
print(253.0 + 238.0 + 248.0)
```
which is a statement that has absolutely nothing to do with `ulab`, yet, it returns 739.0000000000001.

mouryarahul, 2020-11-17:
> @mouryarahul As for issue 6, your problem is that
>
> ```python
> import ulab as np
>
> print(np.__version__)
>
>
> a = np.array([1,2,3], dtype=np.float)
> b = np.array([4,5,6], dtype=np.uint16)
>
> a = [...]

> I was not aware of the fact that this coming from `micropython`. Though it won't create much problem arithmetically in some cases, in some cases, it can. It seems to be a bit inconsistent with its behaviour:

> ```python
> >>> (253.0 + 238.0 + 248.0)
> 739.0000000000001
> >>> (253.0 + 238.0 + 248.0) + (253.0 + 238.0 + 248.0)
> 1478.0
>```

> Sure, we should raise this in `Micropython`

v923z, 2020-11-18:
@mouryarahul I have opened a ticket here: https://github.com/micropython/micropython/issues/6632 It turns out, you don't even need an operation for the odd-looking printout, a single float will do.

mouryarahul, 2020-11-21:
@v923z Unlike CPython, Micropython do not have `math.inf`, but have `math.isinf`. I am not sure if there is any way to create number `inf` in Micropython. Certainly, we do not have `ulab.inf`, but I noticed that some of the function in ulab can create an array with `inf` as elements, e.g., 

> ```python
>  >>> np.lgamma([0, -1, -2, -3, -4])
> array([inf, inf, inf, inf, inf], dtype=float64)
> ```

> Is there a way that an user can create a number `inf` or an array with `inf`?

> Will it be possible to add `np.inf` and `np.nan` like [https://numpy.org/doc/stable/reference/constants.html?highlight=numpy%20inf#numpy.inf](url) ?

v923z, 2020-11-21:
> Will it be possible to add `np.inf` and `np.nan` like [https://numpy.org/doc/stable/reference/constants.html?highlight=numpy%20inf#numpy.inf](url) ?

We could add that, I think this is not hard.

mouryarahul, 2020-11-22:
Inline to the above request, I think it will quite handy for users if they can create an ndarray of  `dtype=ulab.bool` similar to Numpy: 
> ```python
> >>> import numpy as np
> >>> x = np.array([1,0,-1,1,0], dtype=np.bool)
> >>> x
> array([ True, False,  True,  True, False])
> >>> x = np.array([1,0.5,-0.5,1,0], dtype=np.bool)
> >>> x
> array([ True,  True,  True,  True, False])
> >>> x = np.array([1,None,-0.5,1,0], dtype=np.bool)
> >>> x
> array([ True, False,  True,  True, False])
> ```

v923z, 2020-11-22:
> Inline to the above request, I think it will quite handy for users if they can create an ndarray of `dtype=ulab.bool` similar to Numpy:

I am not sure Boolean lists are used too often in this way, but you can always open a new issue ;-0)

https://github.com/v923z/micropython-ulab/issues/252

mouryarahul, 2020-11-22:
> > Inline to the above request, I think it will quite handy for users if they can create an ndarray of `dtype=ulab.bool` similar to Numpy:
> 
> I am not sure Boolean lists are used too often in this way, but you can always open a new issue ;-0)
> 
> #252

@v923z Yeah sure, the examples I showed above are not frequently relevant, but the ability to create an array of boolean is useful in many cases, though the same can be achieved by creating an array of `0` and `1` using `dtype=uint8`.

v923z, 2020-11-23:
@mouryarahul I think all issues have been accounted for, for the two feature requests I have already opened new issues, so this could probably be closed.

mouryarahul, 2020-11-23:
I think this is a great achievement and Thanks! Most of the test scripts what I added didn't work for `circuitpython` (it was crashing at least on my local machine); I don't know actually reasons yet. I hope in the case of `micropython`, the current implementation of ULAB has no issue for now. If so we can close this for now.

closed 2020-11-24 by v923z