v923z/micropython-ulab #232: Raising "Segmentation fault", when use full(), ones() and zeros() method at 3-dimensions mode.

kazuma-kawahara, 2020-11-19:
Hi, thank you for your develop the ulab.

I'm testing 3-dimensions mode.
```
  ulab.h
    #define ULAB_MAX_DIMS  3
```

I get "Segmentation Fault" when use the following methods.
　- full()
　- ones()
　- zeros()

example:
```
>>> import ulab as np
>>>
>>> np.ones((2,5))
Segmentation fault
```

My environment is:
　Ubuntu18.04(micropython/ports/unix)
　micorpython v1.13  (commit 817b80a102a413b6458ea391bb2c463aff99672e)
　ulab v1.10 (commit 531345446fcbace61f5549969a390534aea27e36)
　
Is that methods not supported at the 3-dimensions mode ?

Any help appreciated. Thanks.

v923z, 2020-11-19:
Thanks for the report! This is caused by a trivial error in the code (remnants of the old two-dimensional implementation). I will fix it later today.

v923z, 2020-11-19:
https://github.com/v923z/micropython-ulab/pull/233/ should fix the issue. Here is my test:

```python
import ulab as np

print(np.__version__)

print(np.full((2,2), 3, dtype=np.uint8))
print()
print(np.zeros((2,2,2), dtype=np.uint16))
```
and here is my output

```python
1.4.3-3D-numpy
array([[3, 3],
       [3, 3]], dtype=uint8)

array([[[0, 0],
       [0, 0]],

       [[0, 0],
       [0, 0]]], dtype=uint16)
```

Thanks for taking the time for reporting!

closed 2020-11-19 by v923z