v923z/micropython-ulab #268: Minor sphinx doc warnings

dhalbert, 2020-11-27:
I get these warnings when building docs in CircuitPython with ulab:

```
/home/halbert/repos/circuitpython/shared-bindings/ulab/filter/index.rst:35: WARNING: duplicate object description of ulab.filter.sosfilt, other instance in shared-bindings/ulab/filter/index, use :noindex: for one of them
/home/halbert/repos/circuitpython/shared-bindings/ulab/index.rst:328: WARNING: duplicate object description of ulab.arange, other instance in shared-bindings/ulab/index, use :noindex: for one of them
/home/halbert/repos/circuitpython/shared-bindings/ulab/poly/index.rst:17: WARNING: duplicate object description of ulab.poly.polyfit, other instance in shared-bindings/ulab/poly/index, use :noindex: for one of them
```

v923z, 2020-11-30:
@dhalbert Thanks! I am about to re-organise the code a bit. I'll keep an eye out for this.

v923z, 2021-04-13:
@jepler @dhalbert I think, Jeff fixed this in https://github.com/v923z/micropython-ulab/pull/362. Is that correct?

jepler, 2021-04-14:
I think these particular warnings aren't occurring at the moment.

closed 2021-04-14 by jepler