v923z/micropython-ulab #271: float-conversion compiler-error esp32 mpy1.13-103

CallumJHays, 2020-12-21:
Hello and thank you again for your work on this project. I can't stress how important this is to my project.

I've run into a compiler error when compiling the firmware on master (@c71920e):

```bash
CC /home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c
/home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c: In function 'numerical_median':
/home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c:898:20: error: implicit conversion from 'mp_float_t {aka float}' to 'double' to match other operand of binary expression [-Werror=double-promotion]
             median *= 0.5;
                    ^
/home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c:898:23: error: conversion to 'mp_float_t {aka float}' from 'double' may alter its value [-Werror=float-conversion]
             median *= 0.5;
                       ^
/home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c:934:32: error: implicit conversion from 'mp_float_t {aka float}' to 'double' to match other operand of binary expression [-Werror=double-promotion]
                         median *= 0.5;
                                ^
/home/cal/uni/egh400/bdsim.micropython/firmware-build/ulab/code/numerical/numerical.c:934:35: error: conversion to 'mp_float_t {aka float}' from 'double' may alter its value [-Werror=float-conversion]
                         median *= 0.5;
                                   ^
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:63: build-GENERIC/code/numerical/numerical.o] Error 1
```

I've managed to fix the error by changing all "0.5"s to "0.5f"s, but I'm unsure if this is an adequate or necessary patch.

v923z, 2020-12-23:
@CallumJHays Thanks for reporting the issue! I am not set up for compiling ESP32 at the moment, so I would like to ask you to either submit a patch, or list all your occurrences of this error, so that I can fix it on this side. 

I would also like to suggest to raise an issue at expressif: 0.5 is supposed to be understood as a float, and if that is not the case, that means that they are not conforming to standards. 

And one more request: could you, please, check out the https://github.com/v923z/micropython-ulab/tree/numpy-scipy branch, and see if the errors pop up at the same locations? We could fix that, too. Thanks! 

I would just like to point out that that branch is completely functional, but I didn't have time to fix the docs. Otherwise, it could be released.

CallumJHays, 2020-12-25:
@v923z I've submitted a PR with the fix - this is the only occurrence of this issue in the code-base. I've also tried the `numpy-scipy` branch and the error pops up at the exact same location. Thanks for letting me know about it - I'll be working off that branch from now on.

I've also had a crack at reproducing the standard violation with the ESP32 compiler in a template project, but it appears to be working fine in (relative) isolation.

While I'm at it, I have quite a few numpy-compatability functions that need implementing for my project. Would you prefer that I make a separate issue for each of them as per #76 and periodically send pull requests?

v923z, 2020-12-25:
> @v923z I've submitted a PR with the fix - this is the only occurrence of this issue in the code-base. 

Thanks! I have commented on that. Let me know, how it goes. 

> While I'm at it, I have quite a few numpy-compatability functions that need implementing for my project. Would you prefer that I make a separate issue for each of them as per #76 and periodically send pull requests?

Yes, I think separate issues are much easier to work with. I have a bit of time now, so you could just open a feature request for each function, and we could, in fact, work in parallel then. I would like to ask you to use the `numpy-scipy` branch.

PR #272 (2020-12-27): Specify float literals as float
Fixes #271
closed 2020-12-27 by v923z