v923z/micropython-ulab #273: How to compile micropython-ulab for esp32 board ?

pzg250, 2020-12-30:
Hi,
I am trying to use ESP32_Firmware_v1.12_with_ulab_ulab_v0.54.0_2020-07-29_Generic_PSRAM-4MB_flash-4MB_esp32-spiram_idf4_ulab_sp_thread_v1.12-663-gea4670d5a_2020-07-29.bin, but when I try to test fft, an error appeared. 
Code:
`
x = np.linspace(-np.pi, np.pi, num=8)
`
Output:
```
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  AttributeError: 'module' object has no attribute 'pi'
```
I assume that I use the old version firmware.
Does anyone know how to compile micropython-ulab for esp32 board ? Thanks in advance. 

Regards,

v923z, 2020-12-30:
@pzg250 `pi`, and `e` were added in version 1.0, so no wonder that `ulab` throws an error. You can still do something like this

```
import math
import ulab as np

x = np.linspace(-math.pi, math.pi, num=8)
```

pzg250, 2020-12-31:
@v923z Hi, Thank you very much for your reply. And do we have a guideline that building for esp32? I'd like to use latest version if possible. 

Regards,

pzg250, 2020-12-31:
BTW， seems the pre-build micropython-ulab cannot use bluetooth.

```
import ble_advertising
no module named 'ble_advertising'
```

v923z, 2020-12-31:
> BTW， seems the pre-build micropython-ulab cannot use bluetooth.
> 
> ```
> import ble_advertising
> no module named 'ble_advertising'
> ```

This is an issue for Roberto Colistete's repository, and has nothing to do with `ulab` per se.

v923z, 2020-12-31:
> @v923z Hi, Thank you very much for your reply. And do we have a guideline that building for esp32? I'd like to use latest version if possible.

@pzg250, @mouryarahul was testing `ulab` on an ESP32, so you should ping him.

v923z, 2020-12-31:
@pzg250 You could also look at https://github.com/v923z/micropython-ulab/issues/156, and the references therein.

mouryarahul, 2021-01-04:
@pzg250 Sorry to say that I didn't compile for the ESP32 board. I'm using Micropython PyBoard-D-Series.

pzg250, 2021-01-05:
> @pzg250 Sorry to say that I didn't compile for the ESP32 board. I'm using Micropython PyBoard-D-Series.

Hi @mouryarahul , I think the the compile process is almost same. could you show me the compile steps and command lines ? 
Thanks very much.

Regards,

v923z, 2021-01-05:
> @pzg250 Sorry to say that I didn't compile for the ESP32 board. I'm using Micropython PyBoard-D-Series.

@mouryarahul My bad, sorry!

v923z, 2021-01-05:
@pzg250 Was https://github.com/teuler/robotling/wiki/Adding-native-modules-to-MicroPython-(ESP32)#-adding-ulab-to-the-micropython-firmware of no help? You could ping @teuler, if his notes are not complete.

mdaeron, 2021-01-05:
> And do we have a guideline that building for esp32? I'd like to use latest version if possible.

Back in August I was able to compile `ulab` on a TinyPICO (along with I2S, just comment out the relevant code lines) using this shell script:

````sh
# brew install homebrew/cask/gcc-arm-embedded
# brew install libffi
# brew install pkgconfig

# export PKG_CONFIG_PATH='/usr/local/opt/libffi/lib/pkgconfig'
# export LDFLAGS="-L/usr/local/opt/libffi/lib"
# export CPPFLAGS="-I/usr/local/opt/libffi/include"

clear

echo "\n--- CLONE ULAB ---\n"
git clone https://github.com/v923z/micropython-ulab.git ulab
 
echo "\n--- CLONE MICROPYTHON ---\n"
git clone https://github.com/micropython/micropython.git

cd micropython

echo "\n--- FETCH PR ---\n"
git fetch origin pull/4471/head:esp32_i2s

echo "\n--- MERGE PR ---\n"
git checkout master
git merge esp32_i2s
 
echo "\n--- GIT SUBMODULE UPDATE ---\n"
git submodule update --init
 
echo "\n--- MAKE MPY-CROSS ---\n"
cd mpy-cross
make
 
echo "\n--- MAKE TINYPICO ---\n"
cd ../ports/esp32
export PATH="../../../xtensa-esp32-elf/bin:$PATH"
export ESPIDF=../../../esp-idf
make BOARD=TINYPICO USER_C_MODULES=../../../ulab all
 
echo "\n--- FLASH TINYPICO ---\n"
cd build-TINYPICO
esptool.py --chip esp32 --port /dev/tty.usb* erase_flash
esptool.py --chip esp32 --port /dev/tty.usb* --baud 460800 write_flash -z 0x1000 firmware.bin
````

Maybe this can help? I can't test if it still works right now, but I might be able to later in the week.

v923z, 2021-01-05:
@mdaeron @pzg250 If the script above works, would either of you care to update the https://github.com/v923z/micropython-ulab/blob/master/README.md? Thanks!

pzg250, 2021-01-06:
@mdaeron Thanks very much

pzg250, 2021-01-06:
> @mdaeron @pzg250 If the script above works, would either of you care to update the https://github.com/v923z/micropython-ulab/blob/master/README.md? Thanks!

Sure, I will try to figure it out first

CallumJHays, 2021-01-08:
Hey @pzg250 , If you are still having troubles, this is what I have written up for my personal ESP32 project:

## Building Firmware Manually

These commands assume to be run on linux with access to required dependencies. Please see [The MicroPython README for more information](https://github.com/micropython/micropython#external-dependencies).

### For ESP32

```bash
BUILD_DIR=$(pwd)/firmware # specify build dir
git clone https://github.com/micropython/micropython $BUILD_DIR/micropython
cd $BUILD_DIR/micropython
git checkout b137d064e9e0bfebd2a59a9b312935031252e742
# choose micropython version - note v1.12 is incompatible with ulab
# and v1.13 is currently broken in some ways (on some platforms) https://github.com/BradenM/micropy-cli/issues/167
# - the patch is not live yet (should be in 1.14), but is at this commit
git submodule update --init
cd $BUILD_DIR/micropython/mpy-cross && make # build cross-compiler (required)

cd $BUILD_DIR/micropython/ports/esp32
make ESPIDF= # will display supported ESP-IDF commit hashes
# output should look like: """
# ...
# Supported git hash (v3.3): 9e70825d1e1cbf7988cf36981774300066580ea7
# Supported git hash (v4.0) (experimental): 4c81978a3e2220674a432a588292a4c860eef27b
# """
```

Choose an ESPIDF version from one of the options printed by the previous command:

```bash
ESPIDF_VER=9e70825d1e1cbf7988cf36981774300066580ea7

# Download and prepare the SDK
git clone https://github.com/espressif/esp-idf.git $BUILD_DIR/esp-idf
cd $BUILD_DIR/esp-idf
git checkout $ESPIDF_VER
git submodule update --init --recursive # get idf submodules
pip install -r ./requirements.txt # install python reqs
```

Next, install the ESP32 compiler. If using an ESP-IDF version >= 4.x (chosen by `$ESPIDF_VER` above), this can be done by running `. $BUILD_DIR/esp-idf/install.sh`. Otherwise, (for version 3.x) run:

```bash
cd $BUILD_DIR

# for 64 bit linux
curl https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz | tar xvz

# for 32 bit
# curl https://dl.espressif.com/dl/xtensa-esp32-elf-linux32-1.22.0-80-g6c4433a-5.2.0.tar.gz | tar xvz

# don't worry about adding to path; we'll specify that later

# also, see https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started for more info
```

Next, download `ulab`'s source code and configure required features:

```bash
git clone https://github.com/v923z/micropython-ulab $BUILD_DIR/ulab
cd $BUILD_DIR/ulab
# this commit hash is v1.6.0
git checkout c71920e18636a1e8350cb22d53b119221e0318fc

# you can look through and enable/disable ulab features by editing $BUILD_DIR/ulab/code/ulab.h
# the default enables all features
```

Finally, build the firmware:

```bash
cd $BUILD_DIR/micropython/ports/esp32
# temporarily add esp32 compiler to path
export PATH=$BUILD_DIR/xtensa-esp32-elf/bin:$PATH
export ESPIDF=$BUILD_DIR/esp-idf # req'd by Makefile
export BOARD=GENERIC # options are dirs in ./boards
export USER_C_MODULES=$BUILD_DIR/ulab # include ulab in firmware

make submodules & make all
```

if it compiles without error, you can plug in your ESP32 via USB and then flash it with:

```bash
make erase && make deploy
```

v923z, 2021-01-08:
@CallumJHays Thanks for chipping in! Can you, please, comment on https://github.com/v923z/micropython-ulab/pull/279? This applies to @pzg250, and @mdaeron, too.

closed 2021-01-08 by v923z