v923z/micropython-ulab #274: Add numpy.isinf() & numpy.isfinite()

CallumJHays, 2021-01-08:
`np.isinf` https://numpy.org/doc/stable/reference/generated/numpy.isinf.html

`np.isfinite` https://numpy.org/doc/stable/reference/generated/numpy.isinf.html

v923z, 2021-01-08:
These two functions should be added to https://github.com/v923z/micropython-ulab/blob/master/code/compare/compare.c

v923z, 2021-01-29:
@CallumJHays Cal, would https://github.com/v923z/micropython-ulab/tree/isinf do? I have also added some documentation here: https://github.com/v923z/micropython-ulab/blob/isinf/docs/manual/source/numpy-functions.rst. 

It would be really great, if you could add some tests in https://github.com/v923z/micropython-ulab/tree/isinf/tests. If you drop your test script under `tests/numpy`, and run the `build.sh` script in the root, that should automatically produce the expected results. Let me know if you need help with that.

CallumJHays, 2021-01-30:
@v923z thanks! I was just about to get onto these. I'll try to get them done tomorrow

closed 2021-02-01 by v923z