v923z/micropython-ulab #275: Add numpy ndarray alias?

CallumJHays, 2021-01-08:
Often in my code I check for numpy-arrays via:
```python
if isinstance(x, np.ndarray): ...
```
Because in numpy, `np.array` isn't a class - it's an additional constructor function. `ndarray` is the class.

This is easy to patch in python code but I'm wondering if there'd be merit to exposing `ndarray` as an alias to `ulab.array`?

v923z, 2021-01-08:
Am I correct that you want to define a new class, `ndarray`, and attach all `array` properties and methods to it?

CallumJHays, 2021-01-08:
No, I essentially just want `ndarray = array`

v923z, 2021-01-08:
Would then 
```c
{ MP_OBJ_NEW_QSTR(MP_QSTR_ndarray), (mp_obj_t)&ulab_ndarray_type },
```
do, right after 
https://github.com/v923z/micropython-ulab/blob/9484e846a36bc32253252feca6320c22a56a0839/code/numpy/numpy.c#L49?

CallumJHays, 2021-01-08:
I suppose so? 😅 I'm still not that familiar with the C side of things

v923z, 2021-01-08:
Actually, I think the solution is in your first comment, which I sort of overlooked. 

> Because in `numpy`, `np.array` isn't a class - it's an additional constructor function. `ndarray` is the class.

At the moment, `ulab.numpy.array` is an `ndarray` class. So, I think the solution is to change 

```c
{ MP_OBJ_NEW_QSTR(MP_QSTR_array), (mp_obj_t)&ulab_ndarray_type }, 
```
to 
```c
{ MP_OBJ_NEW_QSTR(MP_QSTR_ndarray), (mp_obj_t)&ulab_ndarray_type }, 
```
and then bind `array` to the constructor, i.e., `ndarray_make_new`. Do you want to take a stab at it, or should I fix it?

CallumJHays, 2021-01-09:
I'll give it a go, looks like a good intro for me into the C code

v923z, 2021-01-09:
The PR looks OK, thanks a lot!

v923z, 2021-01-09:
The issue should be resolved by https://github.com/v923z/micropython-ulab/pull/280.

PR #280 (2021-01-09): Rename array to ndarray, replaced with constructor fn
Fixes #275
closed 2021-01-09 by v923z