v923z/micropython-ulab #277: Add array.__matmul__(), "@" operator

CallumJHays, 2021-01-08:
My understanding is that this functionality is already provided with the `np.linalg.dot` function, but for compatibility's sake could this be supported?

v923z, 2021-01-08:
Yes. The code has to be re-organised a bit, but it is definitely doable.

v923z, 2021-01-30:
@CallumJHays Cal, do you want an implementation that resolves the `dtype`s? At the moment, for reasons of firmware size, `linalg.dot` always returns a `float`, irrespective of the input types. In this regard, the code is not completely `numpy`-compatible, but doing this properly is expensive.

CallumJHays, 2021-01-30:
@v923z I don't think it's necessary. At least for my purposes, floats are the main usecase.

v923z, 2021-02-12:
@CallumJHays Is the @ operator simply an alias for `numpy.dot`, or are there small nuances that we have to keep in mind?

CallumJHays, 2021-02-13:
@v923z I'm not well aware, but [a quick stack-overflow search reveals](https://stackoverflow.com/a/34142617/1266662)

matmul differs from dot in two important ways.
1. Multiplication by scalars is not allowed.
2. Stacks of matrices are broadcast together as if the matrices were elements.

The last point makes it clear that dot and matmul methods behave differently when passed 3D (or higher dimensional) arrays. Quoting from the documentation some more:

>For matmul:
>If either argument is N-D, N > 2, it is treated as a stack of matrices residing in the last two indexes and broadcast accordingly.

>For np.dot:
>For 2-D arrays it is equivalent to matrix multiplication, and for 1-D arrays to inner product of vectors (without complex conjugation). For N dimensions it is a sum product over the last axis of a and the second-to-last of b

I'm working on defining some tests for these differences in terms of code, because that explanation isn't super clear to me. Will post them here when I'm done.

v923z, 2021-02-13:
@CallumJHays You do realise that implementing the generic case is going to be expensive, don't you? Since I am not using this function/method frequently, you have to choose the case that might be most relevant.

thetazero, 2023-03-02:
@v923z Would you be happy with an implementation that just maps `@` to `numpy.linalg.dot`?

v923z, 2023-03-04:
@thetazero You've seen the comments above. Complying with the `numpy` standard might be expensive, and if we go for a partial implementation, I'd like to discuss first which sub-set of the standards we keep. But I would like to see the `@` operator in `ulab`.