v923z/micropython-ulab #278: Add numpy.r_[...] trick (RClass)?

CallumJHays, 2021-01-08:
https://numpy.org/doc/stable/reference/generated/numpy.r_.html

I've managed to contort the original numpy code to get this (mostly) working in micropython, but I'm wondering if this would be worth providing a c implementation? (the numpy implementation is in python). Alternatively, would it be worth it to package a python module with this functionality, or just put it somewhere in the docs incase someone needs it?

Here's my current implementation:
```python
from ulab import array

# this is a stripped down version of RClass (used by np.r_[...etc])
# it doesn't include support for string arguments as the first index element
class RClass:

    def __getitem__(self, key):

        if not isinstance(key, tuple):
            key = (key,)

        objs: List[ndarray] = []
        scalars: List[int] = []
        arraytypes: List[_DType] = []
        scalartypes: List[_DType] = []

        # these may get overridden in following loop
        axis = 0

        for idx, item in enumerate(key):
            scalar = False
            if isinstance(item, slice):
                step = item.step
                start = item.start
                stop = item.stop
                if start is None:
                    start = 0
                if step is None:
                    step = 1
                if isinstance(step, complex):
                    size = int(abs(step))
                    newobj = cast(ndarray, linspace(start, stop, num=size))
                else:
                    newobj = arange(start, stop, step)

            # if is number
            elif isinstance(item, (int, float, complex)):
                newobj = array([item])
                scalars.append(len(objs))
                scalar = True
                scalartypes.append(newobj.dtype())

            else:
                raise Exception("index object %s of type %s is not supported by r_[]" % (
                    str(item), type(item)))

            objs.append(newobj)
            if not scalar and isinstance(newobj, ndarray):
                arraytypes.append(newobj.dtype())

        # Ensure that scalars won't up-cast unless warranted
        # TODO: ensure that this actually works for dtype coercion
        # likelihood is we're going to have to do some funky logic for this
        final_dtype = max(arraytypes + scalartypes)
        if final_dtype is not None:
            for idx in scalars:
                objs[idx] = array(objs[idx], dtype=final_dtype)

        res = concatenate(tuple(objs), axis=axis)

        return res

    # this seems weird - not sure what it's for
    def __len__(self):
        return 0

r_ = RClass()
```

v923z, 2021-02-17:
@CallumJHays I thought that I commented on this issue, but obviously, this is not the case. I would definitely like you to move this piece of code to a more visible place. I was thinking that we could create a `snippets` directory for functions that are implemented in `python`, and this definitely belongs there.

v923z, 2021-03-31:
@CallumJHays Cal, is this OK, or do you want to change something here: https://github.com/v923z/micropython-ulab/pull/358?

CallumJHays, 2021-04-04:
Merged in with #358

closed 2021-04-04 by CallumJHays