v923z/micropython-ulab #282: Anomaly in .tobytes()

lhargrove, 2021-01-11:
I am working with ulab on a Pyboard D-Series SF6. When I use the .tobytes() function, the length seems to be off by a factor of 8. I suspect it is off by a factor of whatever the bytesize of the type is. 

`
>>>
MPY: sync filesystems
MPY: soft reboot
MicroPython v1.13-268-gf7aafc062 on 2021-01-11; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import ulab as np
>>> np.__version__
'1.6.0-2D-numpy'
>>> len(np.zeros((1,1)).flatten().tobytes())
1
>>>

>>>
MPY: sync filesystems
MPY: soft reboot
MicroPython v1.13-268-gf7aafc062 on 2021-01-11; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import ulab as np
>>> len(np.zeros((10,10)).flatten().tobytes())
100
>>>

`

mdaeron, 2021-01-11:
Excellent timing! I was just now about to ask about the following results. I assume thus is not the expected outcome?
````py
from ulab import *

print([x for x in linspace(1,10,10, dtype = int8).tobytes()])
# returns [1, 2, 3, 4, 5, 6, 7, 8, 9, 10]

print([x for x in linspace(1,10,10, dtype = int16).tobytes()])
# returns [1, 0, 2, 0, 3, 0, 4, 0, 5, 0]
````

lhargrove, 2021-01-11:
I think that values are correct, but the returned array is half as large as it should be.

mdaeron, 2021-01-11:
I recompiled `ulab` after changing (in `ndarray.c`, `mp_obj_t ndarray_tobytes()`):

````c
return mp_obj_new_bytearray_by_ref(self->len, self->array);
````

to:

````c
return mp_obj_new_bytearray_by_ref(2*self->len, self->array);
````

This fixes `linspace(1,10,10, dtype = int16).tobytes()` but of course now `linspace(1,10,10, dtype = int8).tobytes()` returns wrong. Now how do I get the correct factor? It's not immediately obvious to me how to access `dtype` from here.

v923z, 2021-01-11:
@mdaeron, @lhargrove can you post the version number? I don't know which version/branch you are compiling.

lhargrove, 2021-01-11:
The version number I am working with is:
'1.6.0-2D-numpy'

v923z, 2021-01-11:
@lhargrove Thanks! I realised after posting my question that you had spelt that out in your original post, that's why I deleted mine. In any case, I will look into it. Thanks for reporting the issue!

v923z, 2021-01-11:
> I recompiled `ulab` after changing (in `ndarray.c`, `mp_obj_t ndarray_tobytes()`):
>
> ```c
> return mp_obj_new_bytearray_by_ref(self->len, self->array);
> ```
>
> to:
>
> ```c
> return [...]

You don't need the `dtype` (although, it is available in the structure), but the `itemsize'. I think, if you change the return value to 

```c
return mp_obj_new_bytearray_by_ref(self->itemsize * self->len, self->array);
```
then it should work. If you manage that, your PR would be welcome.

v923z, 2021-01-11:
@mdaeron @lhargrove I have just checked it, the suggestion above is the fix. Do you want to PR?

mdaeron, 2021-01-11:
> I have just checked it, the suggestion above is the fix. Do you want to PR?

Sure. If you give me a little time I'll look up how to do that.

v923z, 2021-01-11:
@v923z Thanks for the feedback!

closed 2021-01-11 by v923z