v923z/micropython-ulab #284: Unix compilation warnings

mdaeron, 2021-01-12:
Not critical, but I thought I'd point out the following observation: when compiling `ulab` 1.6.1 for unix on a Mac (macOS Catalina), I get two warnings:

````
../../../ulab/code/ulab.c:52:62: error: implicit conversion increases floating-point precision: 'float' to 'mp_float_t' (aka 'double') [-Werror,-Wdouble-promotion]
mp_obj_float_t ulab_const_float_inf_obj = {{&mp_type_float}, INFINITY};
                                          ~                  ^~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/math.h:68:21: note: expanded from macro 'INFINITY'
#define INFINITY    HUGE_VALF
                    ^~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/math.h:58:28: note: expanded from macro 'HUGE_VALF'
#   define    HUGE_VALF    __builtin_huge_valf()
                           ^~~~~~~~~~~~~~~~~~~~~
../../../ulab/code/ulab.c:56:62: error: implicit conversion increases floating-point precision: 'float' to 'mp_float_t' (aka 'double') [-Werror,-Wdouble-promotion]
mp_obj_float_t ulab_const_float_nan_obj = {{&mp_type_float}, NAN};
                                          ~                  ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/math.h:60:28: note: expanded from macro 'NAN'
#   define    NAN          __builtin_nanf("0x7fc00000")
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
````

I can suppress these warnings by appending `CFLAGS += -Wno-double-promotion` to `micropython/ports/unix/MakeFile`. The resulting binary appears to run just fine.

I don't know how to correct the underlying issue, but if someone addresses it I can test whether the warnings disappear.

v923z, 2021-01-12:
@mdaeron I believe, you can get rid of the warning by making the conversion explicit with `MICROPY_FLOAT_CONST(INFINITY)`. If that is the case, we should definitely patch up the code.

mdaeron, 2021-01-12:
Maybe I misunderstood but I replaced:
````
mp_obj_float_t ulab_const_float_inf_obj = {{&mp_type_float}, INFINITY};
````
with:
````
mp_obj_float_t ulab_const_float_inf_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(INFINITY)};
````
and:
````
mp_obj_float_t ulab_const_float_nan_obj = {{&mp_type_float}, NAN};
````
with:
````
mp_obj_float_t ulab_const_float_nan_obj = {{&mp_type_float}, MICROPY_FLOAT_CONST(NAN)};
````
And the warning is not affected.

v923z, 2021-01-13:
> Maybe I misunderstood but I replaced:

No, that is what I actually meant. I couldn't test it, not having a mac.

v923z, 2021-01-13:
@mdaeron I forgot to ask, do you get a similar warning, if you compile for other ports? If this is a generic issue, then we can add your fix to the make fragment in https://github.com/v923z/micropython-ulab/blob/master/code/micropython.mk.

mdaeron, 2021-01-13:
> @mdaeron I forgot to ask, do you get a similar warning, if you compile for other ports? If this is a generic issue, then we can add your fix to the make fragment in https://github.com/v923z/micropython-ulab/blob/master/code/micropython.mk.

I didn't get that warning when compiling for TinyPICO a few days ago. I'll try to recompile for the pyboard now and report here.

v923z, 2021-01-13:
> I didn't get that warning when compiling for TinyPICO a few days ago. I'll try to recompile for the pyboard now and report here.

Then it is most probably specific to the `unix` port. I think we should keep the issue open.

v923z, 2021-01-13:
@mdaeron If you still feel like trying things out, what happens, if you have this:

```c
mp_obj_float_t numpy_const_float_inf_obj = {{&mp_type_float}, (mp_float_t)INFINITY};
```

mdaeron, 2021-01-13:
> @mdaeron If you still feel like trying things out, what happens, if you have this:
> 
> ```c
> mp_obj_float_t numpy_const_float_inf_obj = {{&mp_type_float}, (mp_float_t)INFINITY};
> ```

I get the following errors:
````
CC ../../../ulab/code/ulab.c
../../../ulab/code/ulab.c:405:49: error: use of undeclared identifier 'ulab_const_float_inf_obj'
        { MP_ROM_QSTR(MP_QSTR_inf), MP_ROM_PTR(&ulab_const_float_inf_obj) },
                                                ^
../../../ulab/code/ulab.c:408:49: error: use of undeclared identifier 'ulab_const_float_nan_obj'
        { MP_ROM_QSTR(MP_QSTR_nan), MP_ROM_PTR(&ulab_const_float_nan_obj) },
                                                ^
../../../ulab/code/ulab.c:419:8: error: invalid application of 'sizeof' to an incomplete type 'const mp_map_elem_t []'
STATIC MP_DEFINE_CONST_DICT (
       ^~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:374:21: note: expanded from macro 'MP_DEFINE_CONST_DICT'
            .used = MP_ARRAY_SIZE(table_name), \
                    ^~~~~~~~~~~~~~~~~~~~~~~~~
../../py/misc.h:114:33: note: expanded from macro 'MP_ARRAY_SIZE'
#define MP_ARRAY_SIZE(a) (sizeof(a) / sizeof((a)[0]))
                                ^~~
../../../ulab/code/ulab.c:419:8: error: invalid application of 'sizeof' to an incomplete type 'const mp_map_elem_t []'
STATIC MP_DEFINE_CONST_DICT (
       ^~~~~~~~~~~~~~~~~~~~~~
../../py/obj.h:375:22: note: expanded from macro 'MP_DEFINE_CONST_DICT'
            .alloc = MP_ARRAY_SIZE(table_name), \
                     ^~~~~~~~~~~~~~~~~~~~~~~~~
../../py/misc.h:114:33: note: expanded from macro 'MP_ARRAY_SIZE'
#define MP_ARRAY_SIZE(a) (sizeof(a) / sizeof((a)[0]))
                                ^~~
````

v923z, 2021-01-13:
This is really mean. On `linux` the code above compiles without errors.

v923z, 2021-01-14:
@mdaeron I think https://github.com/v923z/micropython-ulab/pull/286 solves this issue. Re-open it, if you still have difficulties!

closed 2021-01-14 by v923z

mdaeron, 2021-01-14:
I confirm this fixes the problem.

v923z, 2021-01-14:
Thanks for the feedback!

commit 00e07bfa (2021-01-14): fix clang cast problem

Closes #284.