v923z/micropython-ulab #285: Issue with np.diff()

lhargrove, 2021-01-13:
I was working with the np.diff() function on the Pyboard D Series SF6. I noticed that the diff() function doesn't seem to be returning what I expect. I am using version 1.6.0-2D-numpy.  After I noticed the issue, I cut and paste values from the documentation and saw some errors as well: 

`
MicroPython v1.13-268-gf7aafc062-dirty on 2021-01-11; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
import ulab as np
a = np.array(range(9), dtype=np.uint8)
print('a:\n', a)
print('\nfirst derivative:\n', np.diff(a, n=1))
print('\nsecond derivative:\n', np.diff(a, n=2))
c = np.array([[1, 2, 3, 4], [4, 3, 2, 1], [1, 4, 9, 16], [0, 0, 0, 0]])
print('\nc:\n', c)
print('\nfirst derivative, first axis:\n', np.diff(c, axis=0))
print('\nfirst derivative, second axis:\n', np.diff(c, axis=1))
a:
 array([0, 1, 2, 3, 4, 5, 6, 7, 8], dtype=uint8)

first derivative:
 array([1, 1, 1, 1, 1, 1, 1, 1], dtype=uint8)

second derivative:
 array([0, 0, 0, 0, 0, 0, 0], dtype=uint8)

c:
 array([[1.0, 2.0, 3.0, 4.0],
       [4.0, 3.0, 2.0, 1.0],
       [1.0, 4.0, 9.0, 16.0],
       [0.0, 0.0, 0.0, 0.0]], dtype=float64)

first derivative, first axis:
 array([[3.0, 1.0, -1.0, 4.0],
       [3.0, 1.0, -8.911015922500841e+252, 4.0],
       [3.0, -2.504798098753954e+262, -8.911015922500841e+252, 4.0]], dtype=float64)

first derivative, second axis:
 array([[1.0, -1.0, 3.0],
       [2.504798098753954e+262, 1.0, -1.0],
       [1.0, 2.504798098753954e+262, 1.0],
       [-4.788098201154537e-299, 1.0, 2.504798098753954e+262]], dtype=float64)
>>>
`

v923z, 2021-01-14:
@lhargrove Thanks for reporting the issue! I have just noticed this error, too. I am working on a fix, should be done later today.

v923z, 2021-01-14:
@lhargrove I think https://github.com/v923z/micropython-ulab/pull/287 should solve the problem.

closed 2021-01-14 by v923z