v923z/micropython-ulab #290: numpy.array() can't create 3-dimensional array.

kazuma-kawahara, 2021-01-18:
Hi, thank you for your develop the ulab.

I'm testing 3-dimensions mode.
```
  ulab.h
    #define ULAB_MAX_DIMS  3
```

I used numpy.array() to create 3-dimensional array, but error is ocurred.
```
>>> from ulab import numpy as np
>>> np.array([[[1,2],[3,4]],[[5,6],[7,8]]])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert list to float
```

Is that methods not supported at the 3-dimensions mode ?

Any help appreciated. Thanks.

v923z, 2021-01-18:
@kazuma-kawahara This is not spelt out in the manual, but direct initialisation from lists is supported only for 1 and 2D. You can do 
```
np.array(range(8)).reshape((2,2,2)) + 1
```
but you probably want to have something more general.

In any case, we can look into adding what you need. It's is not terribly difficult.

kazuma-kawahara, 2021-01-19:
Thank you for your reply and your consideration.

As a side note, I use "reshape ()" or "concatenate ()" to generate 3D arrays.

Thank you.

v923z, 2021-01-29:
@kazuma-kawahara I think https://github.com/v923z/micropython-ulab/pull/297 should address your question. I am closing the issue, if you run into difficulties, please, open a new one.

closed 2021-01-29 by v923z