v923z/micropython-ulab #292: Activating the user modul fails

teuler, 2021-01-22:
Hi, 
I am trying to include the standard user module (`user/user.c`) into ulab.
I set `#define ULAB_HAS_USER_MODULE` to `(1)` in `ulab.h` and recompiled, basically following the instructions in ulab-readme file. It successfully generates the firmware (`user.c` turns up in the list of compiled files) but when I try `from ulab import user` it throws an error (`ImportError: can't import name user`). Also `dir(ulab)` does not show the user module. The other `ulab` modules work fine. 

What am I missing? Last year, I added user functions a couple of times and it always worked. 

Thanks
Thomas

ESP32, newest version of MicroPython, newest version of `ulab`

teuler, 2021-01-23:
... I think I found the problem. More soon.

v923z, 2021-01-24:
@teuler I merged your PR, I think we can close the issue now.

closed 2021-01-24 by v923z