v923z/micropython-ulab #294: ulab.numpy.array objects .shape not complete compatible numpy.array objects .shape

water5, 2021-01-23:
ulab.numpy.array objects .shape -> <bound_method>
ulab.numpy.array objects .shape() -> <class 'tuple'>

numpy.array objects .shape -> <class 'tuple'>

v923z, 2021-01-24:
The problem is that, as far as I know, defining properties is not implemented in `micropython`, so `.shape()` is really just a workaround. I will try to mount an effort, and get my PR concerning this issue accepted in `micropython` proper.

jgpeiro, 2021-02-13:
But micropython DOES support properties, right?( I wanted to show you a repo that I remember about how to implement them in C...but its yours too!) [https://github.com/v923z/micropython-usermod/blob/master/snippets/properties/properties.c](url)

v923z, 2021-02-13:
@jgpeiro The repo that you link to discusses a simple case, but here we would need something more complex: https://github.com/micropython/micropython/pull/5648

water5, 2021-07-14:
Properties implemented.

closed 2021-07-14 by water5