v923z/micropython-ulab #296: ESP32 port speed deterioration

mtk11, 2021-01-27:
W/o micropython-ulab:

Pystone(1.2) time for 500 passes = 465ms
This machine benchmarks at 1075 pystones/second

With micropython-ulab 2.1.3:

Pystone(1.2) time for 500 passes = 927ms
This machine benchmarks at 539 pystones/second

What can be causing the speed decrease ?

v923z, 2021-01-27:
@mtk11 Can you point out, where `pystone` would refer to `ulab`? I presume, you are using this https://github.com/micropython/micropython-lib/blob/master/pystone/pystone.py, but I don't see, where `ulab` would play any role here.

mtk11, 2021-01-27:
@v923z Using pystone_lowmem:  https://github.com/micropython/micropython-lib/blob/master/pystone_lowmem/pystone_lowmem.py

In ulab.h, just left the following (apparently the minimum for compiling):

#define ULAB_NUMPY_HAS_ARGMINMAX        (1)
#define ULAB_NUMPY_HAS_MEAN             (1)
#define ULAB_NUMPY_HAS_SUM              (1)

Got some warnings during the building:

optimize.c:28:19: warning: 'optimize_python_call' defined but not used [-Wunused-function]
 static mp_float_t optimize_python_call(const mp_obj_type_t *type, mp_obj_t fun, mp_float_t x, mp_obj_t *fargs, uint8_t nparams) {
….
compare.c:121:17: warning: 'compare_equal_helper' defined but not used [-Wunused-function]
 static mp_obj_t compare_equal_helper(mp_obj_t x1, mp_obj_t x2, uint8_t comptype) {
….
vector.c:36:17: warning: 'vectorise_generic_vector' defined but not used [-Wunused-function]
 static mp_obj_t vectorise_generic_vector(mp_obj_t o_in, mp_float_t (*f)(mp_float_t)) {

The Pystone result is:

Pystone(1.2) time for 500 passes = 643ms
This machine benchmarks at 777 pystones/second

By the way, in regards to add numpy.cov #188, any plans to add also [numpy.correlate](https://numpy.org/doc/stable/reference/generated/numpy.correlate.html#numpy.correlate) ?

v923z, 2021-01-27:
> @v923z Using pystone_lowmem: https://github.com/micropython/micropython-lib/blob/master/pystone_lowmem/pystone_lowmem.py
> 
> In ulab.h, just left the following (apparently the minimum for compiling):
> 
> #define ULAB_NUMPY_HAS_ARGMINMAX (1)
> #define ULAB_NUMPY_HAS_MEAN (1)
> #define ULAB_NUMPY_HAS_SUM (1)

You should be able to disable functions arbitrarily. If that is not the case then some of the pre-processor switches are not in the code itself.  

> Got some warnings during the building:
>
> optimize.c:28:19: warning: 'optimize_python_call' defined but not used [-Wunused-function]
> static mp_float_t optimize_python_call(const mp_obj_type_t *type, [...]

Here I assume that you removed basically all modules. These three functions are somewhat generic, and are used by many functions, that is why there is no #if/#endif clause around them, hence the warnings. I can fix it. 

> The Pystone result is:
> 
> Pystone(1.2) time for 500 passes = 643ms
> This machine benchmarks at 777 pystones/second

The `pystone` code that you linked to has absolutely no reference to `ulab`. I am really confused here, because how should that piece of code know, which modules you have? Is it possible that by including `ulab`, you pass some flash boundary, and then the core functions can no longer load fast enough? Can you include some other module instead of `ulab` to see, whether that is the case?

To be honest, I don't even know at the moment, where to look for the cause of the problem. 

> By the way, in regards to add numpy.cov #188, any plans to add also [numpy.correlate](https://numpy.org/doc/stable/reference/generated/numpy.correlate.html#numpy.correlate) ?

I don't have too much time, but I keep this in mind. Alternatively, contributions are always welcome. If you need help with that, just let me know.

mtk11, 2021-01-28:
@v923z 

> The `pystone` code that you linked to has absolutely no reference to `ulab`. I am really confused here, because how should that piece of code know, which modules you have? Is it possible that by [...]

Removed some of the ESP32 port components, which decreased the total firmware size to 1314 kB and this restored the speed. Presume the latest ESP32 micropython port reached the flash size limitations.

closed 2021-01-28 by mtk11