v923z/micropython-ulab #301: Constructed array using ulab.numpy.ndarray()

water5, 2021-02-01:
[https://numpy.org/doc/stable/reference/generated/numpy.ndarray.html](https://numpy.org/doc/stable/reference/generated/numpy.ndarray.html)

Refer numpy.ndarray(shape), its parameter is shape, although arrays should be constructed using array(), arange(), zeros() ... and ulab.numpy support 2D array currently, but I think compatible numpy.ndarray() will be a good idea.

Currently:
`ulab.numpy.ndarray((2, 2))` -> array([2.0, 2.0], dtype=float32)

v923z, 2021-02-01:
This is correct, but I don't see at the moment, where the problem is. I am afraid, it might have to do with how `micropython` is implemented. I will look into this, though.

CallumJHays, 2021-02-01:
Was this solved by #280? That push results in the following code:
```

CallumJHays, 2021-02-01:
Ah, looks like I didn't read the spec properly for #280 :sweat_smile: 
I'll try to get a fix out by tomorrow.

v923z, 2021-02-02:
@CallumJHays OK, thanks! I think you are right, one can catch this in the constructor function, or in `ndarray_make_new`.