v923z/micropython-ulab #304: ulab.numpy.zeros((0, n)) will cause MicroPython crash.

water5, 2021-02-02:
Although it has nothing implication, but it should be check before run it.

`np.zeros((0, 3))`

> array([Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
>
> Core 0 register dump:
> PC      : 0x400eabf1  PS      : 0x00060830  A0      : 0x801030cd  A1      : 0x3ffce980  
> A2      : 0x00000066  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x0000ff00  
> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000066  A9      : 0x3ffce950  
> A10     : 0x00000003  A11     : 0x3f407632  A12     : 0x00000001  A13     : 0x00000002  
> A14     : 0x3ffceb34  A15     : 0x00000004  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000000  LBEG    : 0x40095bfd  LEND    : 0x40095c0d  LCOUNT  : 0xfffffffe  
>
> ...
>
> Rebooting...

v923z, 2021-02-04:
> Although it has nothing implication, but it should be check before run it.
> 
> `np.zeros((0, 3))`

While it is definitely a corner case, I think this still has implications, and should be caught. You might not know beforehand, what the shape of your tensor is.

water5, 2021-02-04:
Append, on CircuitPython does not cause CircuitPython crash, but array values incorrect.
Test environment: CircuitPython 6.1.0 Release, STM32F405RGT6.

`ulab.zeros((0, 3))`
array([1.10114e-19, 3.99414e-34, 3.99417e-34], dtype=float32)

v923z, 2021-02-06:
> Test environment: CircuitPython 6.1.0 Release, STM32F405RGT6.
> 
> `ulab.zeros((0, 3))`
> 
> > array([1.10114e-19, 3.99414e-34, 3.99417e-34], dtype=float32)
> 
> Test environment: MicroPython 1.13, STM32.
> 
> `ulab.numpy.zeros((0, 3))`
> 
> > array([2.533597e-29, 3.9553e-34, 3.937858e-34], dtype=float32)

I think both examples just show that the array is uninitialised, though the shape is incorrect

v923z, 2021-02-08:
@water5 Can you link an overview of empty arrays in `numpy`?

water5, 2021-02-11:
@v923z Empty arrays? how?
Like this?
`np.zeros((0, ))`

> array(, dtype=float32)

`np.zeros((0, 0))`

> array(, dtype=float32)

v923z, 2021-02-12:
@water5 In `numpy`, `zeros((0, 3))`, your original example, produces an empty array `array([], shape=(0, 3), dtype=float64)`. What is not clear to me, what this means. What happens, if you operate on such an array? E.g., 

```
sum(zeros((0, 3)))
```
```
0.0
```
but
```
std(zeros((0, 3)))
```
```
nan
```
and 
```
std([1.0])
```
```
0.0
```

What is the rationale here? 
How is broadcasting resolved, whether you can matrix multiply  two empty arrays, and if you can, what is the result, etc? I haven't found any discussion on this subject, and have the feeling that this is a corner case even in `numpy`.

If you ask for a fix, then you have to tell us, what the expected behaviour is. I understand that the function shouldn't reboot `micropython`, and https://github.com/v923z/micropython-ulab/pull/310 prevents that, but I presume, that is not the full story, and this is why I asked for some background information.

v923z, 2021-05-20:
@water5 I think this is solved in https://github.com/v923z/micropython-ulab/pull/392, so closing now.

closed 2021-05-20 by v923z