v923z/micropython-ulab #307: How to perform vector  2-norm?

ubIQio, 2021-02-07:
I have a set of 3-D acceleration vectors that I'd like to convert to single values(ie the square root of the sum of squared acceleration in x, y, and z ) so that I can compute a spectogram of the total acceleration

In numpy I can do 
``` python
import numpy as np

# create sample array
a = np.arange(12).reshape((4,3))
print('\nsample array\n',a)

# compute vector norms
print('\n2-norm',np.linalg.norm(a, axis=1))
```
to get

```python
sample array
 [[ 0  1  2]
 [ 3  4  5]
 [ 6  7  8]
 [ 9 10 11]]

2-norm [ 2.23606798  7.07106781 12.20655562 17.3781472 ]
```

What would be the recommended way to do this in ulab?

v923z, 2021-02-07:
You simply need the square root of the sum of the squares, right? Would this work?

```
a = np.arange(12).reshape((4,3))

[sqrt(sum(a[i]*a[i])) for i in a.shape()[0]]
```

ubIQio, 2021-02-07:
Well it works, but not in my application  :(

```python
a = np.arange(1024*3,dtype=np.uint16).reshape((1024,3))

@timeit 
def n(a):
    return [sqrt(sum(a[i]*a[i])) for i in range(a.shape()[0])]

b=n(a)

```

yields

``` python
execution time:  218432  us
```

At the rate the data is arriving, I just don't have the time budget for that.

And the 1024-point spectrogram that follows only takes 3ms; I'm afraid I am spoilt by ulab.

Using np.linalg.norm(a) only takes ~1ms, but then only 1 sqrt()...

As a test,  I modified np.linalg.norm() to do the MICROPY_FLOAT_C_FUN(sqrt)(dot) on every row, ie 1024 times in my case, and it only takes 1.6ms, but I haven't modified it to output the right shape.

Any thoughts?

v923z, 2021-02-08:
We can add the `axis` keyword argument, if you need it. What makes your code run slow is the huge number of function calls. Depending on which platform you are on, a single call can take tens of microseconds, which then adds up to the 200 ms that you mentioned. 

However, in the meantime, I _believe_, you can still do something like this: 

```
_std = np.std(a, axis=1)
_mean = np.mean(a, axis=1)
_norm = np.sqrt(_std**2 + _mean**2)
```
(You have to work out the exact math.) These are three function calls, and two binary operations. This should save you some time. 

> And the 1024-point spectrogram that follows only takes 3ms;

If I understand you correctly, you take the Fourier transform here. But if it so, can't you take the transform of x, y, and z, and sum the squares of the Fourier components at the end? It shouldn't matter, where you take the norm, in real space, or in Fourier space. In the worst case, you'll be off by a constant multiplicative factor, which you probably don't care about. 

```
sx = np.scipy.signal.spectrogram(x)
sy = np.scipy.signal.spectrogram(y)
sz = np.scipy.signal.spectrogram(z)

sall = np.sqrt(sx**2 + sy**2 + sz**2)
# if you need only the maximum, or something like that, then you can leave out the square root
sall = sx**2 + sy**2 + sz**2
```

v923z, 2021-02-09:
@ubIQio The latest version implements the `axis` keyword for `linalg.norm` https://github.com/v923z/micropython-ulab/pull/312. Do you think you could update the test script in https://github.com/v923z/micropython-ulab/blob/master/tests/numpy/linalg.py? People, who contribute, usually get brownie points: https://github.com/v923z/micropython-ulab/graphs/contributors :wink:

ubIQio, 2021-02-10:
Wow!

The test below works, IF the tolerance is worse than 1E-7

```python
a = np.array([[0, 1, 2], [5, 4, 8], [4, 4, 8] ], dtype=np.int16)
result = (np.linalg.norm(a,axis=1))
ref_result = np.array([2.236068, 10.24695, 9.797959])
for i in range(3):
        print(math.isclose(result[i], ref_result[i], rel_tol=1E-6, abs_tol=1E-6))
```
however if the tolerance is greater, eg 1E-9, the test seems to fail if  a resulting array element  is above 10, in this case a[1]

But this is excellent progress, and fyi the execution time on my platform is 3ms for 1024 3D vectors.

I am sorry I am not sure how to integrate this into your test scripts.

Thank you so much

v923z, 2021-02-12:
https://github.com/v923z/micropython-ulab/pull/314 fixes the issue, so closing now.

closed 2021-02-12 by v923z