v923z/micropython-ulab #308: installing 'ulab': (-17040, 'MBEDTLS_ERR_RSA_PUBLIC_FAILED+MBEDTLS_ERR_MPI_ALLOC_FAILED'), packages may be partially installed

caxefaizan, 2021-02-08:
I'm new to Micropython and want to use this library for some calculations. I tried to install the library in uPyCraft V1.1 using the command `pip.install("ulab")` . But am facing this error! Am i doing it wrong?

v923z, 2021-02-08:
I think, with `upip`, you can install only `micropython` packages. If you want `ulab` to be part of your firmware, you have to compile from the source: https://github.com/v923z/micropython-ulab/#firmware

caxefaizan, 2021-02-08:
> I think, with `upip`, you can install only `micropython` packages. If you want `ulab` to be part of your firmware, you have to compile from the source: https://github.com/v923z/micropython-ulab/#firmware

makes sense... can you hook me up with a tutorial on how to do that? i'm a newbie in this case

v923z, 2021-02-08:
> makes sense... can you hook me up with a tutorial on how to do that? i'm a newbie in this case

You should simply follow the steps in that section of the readme.

caxefaizan, 2021-02-08:
> > makes sense... can you hook me up with a tutorial on how to do that? i'm a newbie in this case
> 
> You should simply follow the steps in that section of the readme.

Okay so i downloaded the `firmware`  `ESP32_Firmware_v1.12_with_ulab_ulab_v0.54.0_2020-07-29_Generic_flash-4MB_esp32_idf3_ulab_sp_thread_v1.12-663-gea4670d5a_2020-07-29` for my Adafruit Huzzah 32  from gitlab. and now i can `import ulab` in the terminal. however i am getting the error `AttributeError: 'module' object has no attribute 'ndarray'` when i run the command `a = ulab.ndarray([1, 2, 3, 4, 5], dtype=ulab.uint8)` as shown in the example.

caxefaizan, 2021-02-08:
> Okay so i downloaded the `firmware` `ESP32_Firmware_v1.12_with_ulab_ulab_v0.54.0_2020-07-29_Generic_flash-4MB_esp32_idf3_ulab_sp_thread_v1.12-663-gea4670d5a_2020-07-29` for my Adafruit Huzzah 32 [...]

Okay!!! Got it .... used `dir(ulab)` to find out the methods available and found it to be `a = ulab.array([1, 2, 3, 4, 5], dtype=ulab.uint8)` !!! Thanks

closed 2021-02-08 by caxefaizan

v923z, 2021-02-08:
> > Okay so i downloaded the `firmware` `ESP32_Firmware_v1.12_with_ulab_ulab_v0.54.0_2020-07-29_Generic_flash-4MB_esp32_idf3_ulab_sp_thread_v1.12-663-gea4670d5a_2020-07-29` for my Adafruit Huzzah 32 from gitlab. 

This is a very old version, a lot of things have since changed. 

> Okay!!! Got it .... used `dir(ulab)` to find out the methods available and found it to be `a = ulab.array([1, 2, 3, 4, 5], dtype=ulab.uint8)` !!! Thanks

Well, you have found the solution yourself.