v923z/micropython-ulab #315: possible bug in...ulab dot/transpose??

jgpeiro, 2021-02-11:
I am trying to implement the kalman filter [https://github.com/zziz/kalman-filter](url) and I have seen some strange behavior in the libraries that I cant explain.
When I perform the multiplications inside the kalman.predict method the results are different from what I obtain in the "real numpy" version. However, when I perform the operations outside this function the results are correct.
At first I thought it was a precision problem (float vs double), but it seems like a lot of error for float operations, also the error would appear in both tests

```
import time
import ulab
from ulab import numpy as np

np_dot = np.linalg.dot
print( "version", ulab.__version__ )

class KalmanFilter:
    def __init__(self, F = None, B = None, H = None, Q = None, R = None, P = None, x0 = None):
        if(F is None or H is None):
            raise ValueError("Set proper system dynamics.")
        self.n = F.shape()[1]
        self.m = H.shape()[1]
        self.F = F
        self.H = H
        self.B = np.zeros(1, dtype=np.float) if B is None else B
        self.Q = np.eye(self.n, dtype=np.float) if Q is None else Q
        self.R = np.eye(self.n, dtype=np.float) if R is None else R
        self.P = np.eye(self.n, dtype=np.float) if P is None else P
        self.x = np.zeros((self.n, 1), dtype=np.float) if x0 is None else x0

    def predict(self, u ):
        self.x = np_dot( self.F, self.x ) + np_dot( self.B, u )

        a = np_dot( self.F, self.P )
        print( "a", a )
        b = self.F.transpose()
        print( "b", b )
        c = np_dot( a, b )        
        print( "c", c )
        
        self.P = c + self.Q
        return np_dot( H, self.x )[0]

    def update(self, z):
        y = z - np_dot( self.H, self.x )
        S = self.R + np_dot( self.H, np_dot( self.P, self.H.transpose() ) )
        K = np_dot( np_dot( self.P, self.H.transpose() ), np.linalg.inv(S) )
        self.x = self.x + np_dot( K, y )
        I = np.eye( self.n )
        self.P = np_dot( np_dot( I - np_dot( K, self.H ), self.P ), 
            (I - np_dot( K, self.H ) ).transpose() ) + np_dot( np_dot( K, self.R ), K.transpose() )

dt = 1.0/100
points = 100

F = np.array( [[1, dt, 0], [0, 1, dt], [0, 0, 1]] )
H = np.array( [1, 0, 0] ).reshape( (1, 3) )
Q = np.array( [[0.05, 0.05, 0.0], [0.05, 0.05, 0.0], [0.0, 0.0, 0.0]] )
R = np.array( [0.5] ).reshape( (1, 1) )
u = np.array( [0] )

x = np.linspace(-10, 10, points)
real = -(x**2 + 2*x - 2)
#noise = np.random.normal(0, 2, 100)
measurements = real# + noise

kf = KalmanFilter(F = F, H = H, Q = Q, R = R)
predictions = np.zeros( points )

print( "kalman.predict" )
for i in range( 1 ):
    m = measurements[i]
    kf.update( m )
    kf.predict( u )

print("main")
a = np.array([[0.3333333, 0.01, 0.0],
       [0.0, 1.0, 0.01],
       [0.0, 0.0, 1.0]] )
print( "a", a )

b = np.array([[1.0, 0.0, 0.0],
       [0.01, 1.0, 0.0],
       [0.0, 0.01, 1.0]] )
print( "b", b )

c = np_dot( a, b )
print( "c", c )
```

OUTPUT:

```
version 2.3.4-2D
kalman.predict
a array([[0.3333333, 0.01, 0.0],
       [0.0, 1.0, 0.01],
       [0.0, 0.0, 1.0]], dtype=float32)
b array([[1.0, 0.0, 0.0],
       [0.01, 1.0, 0.0],
       [0.0, 0.01, 1.0]], dtype=float32)
c array([[0.3333333, 0.01333333, 0.0001],    <<<<< this result looks worng
       [0.0, 1.0, 0.02],
       [0.0, 0.0, 1.0]], dtype=float32)
main
a array([[0.3333333, 0.01, 0.0],
       [0.0, 1.0, 0.01],
       [0.0, 0.0, 1.0]], dtype=float32)
b array([[1.0, 0.0, 0.0],
       [0.01, 1.0, 0.0],
       [0.0, 0.01, 1.0]], dtype=float32)
c array([[0.3334333, 0.01, 0.0],             <<<<< this result matches the real numpy version
       [0.01, 1.0001, 0.01],
       [0.0, 0.01, 1.0]], dtype=float32)
```

v923z, 2021-02-11:
@jgpeiro Do you think you can separate the `ulab` code from your class? It is a bit hard to understand what is going on, if we have to deal with an extra layer of abstraction. A minimal example demonstrating the issue would be great. Something like

```
a = array([[1, 2], [3, 4]])
print(a.transpose())
```
```
array([[45, 12], [25, 99]], dtype=float32)
```

Thanks!

jgpeiro, 2021-02-11:
I know. I post with kalman code cause I wasent sure where the problem comes from... but here is a clear version:

```
from ulab import numpy as np

np_dot = np.linalg.dot

a = np.array( [
    [0.3333333, 0.01, 0.0],
    [0.0, 1.0, 0.01],
    [0.0, 0.0, 1.0]
], dtype=np.float )

b = np.array( [
    [1.0, 0.01, 0.0],
    [0.0, 1.0, 0.01],
    [0.0, 0.0, 1.0]
], dtype=np.float )

c = np_dot( a, b.transpose() )

print( "c", c )
```
Output:
```
c array([[0.3333333, 0.01333333, 0.0001],
       [0.0, 1.0, 0.02],
       [0.0, 0.0, 1.0]], dtype=float32)
```

This is the code on "real numpy"
```
import numpy as np

np_dot = np.dot

a = np.array( [
    [0.3333333, 0.01, 0.0],
    [0.0, 1.0, 0.01],
    [0.0, 0.0, 1.0]
], dtype=np.float32 )

b = np.array( [
    [1.0, 0.01, 0.0],
    [0.0, 1.0, 0.01],
    [0.0, 0.0, 1.0]
], dtype=np.float32 )

c = np_dot( a, b.transpose() )

print( "c", c )
```
Output:
```
c [[0.3334333 0.01      0.       ]
 [0.01      1.0001    0.01     ]
 [0.        0.01      1.       ]]
```

v923z, 2021-02-12:
@jgpeiro I see that you have to define `np_dot`, because in `numpy` `dot` is at the top level, and not in the `linalg` module. Is that correct? I don't even remember, why I added the function to `linalg`...

jgpeiro, 2021-02-12:
Yes, I saw this and I thought that maybe in some numpy version dot came inside linalg module...so I simply copy to retain code more understandable... but I knto that move from one package to another will be an easy task. 
I open the issue cause Im worried about transpose function. This is clearly a problem somewhere, or Im wrong?
(For know I fix with "b.transpose()+0" that I think forces new array allocation)
```
from ulab import numpy as np
np_dot = np.linalg.dot
a = np.array( [
    [1,2],
    [3,4]
], dtype=np.float )

b = np.array( [
    [5,6],
    [7,8]
], dtype=np.float )

print( np_dot( a, b.transpose() ) )

Ulab Output:
[[19.0, 22.0],
 [43.0, 50.0]]

Expected:
[[17. 23.]
 [39. 53.]]
```

v923z, 2021-02-12:
> I open the issue cause Im worried about transpose function. This is clearly a problem somewhere, or Im wrong?
> (For know I fix with "b.transpose()+0" that I think forces new array allocation)

Oh, I see. Indeed, the `.transpose()` method simply re-writes the `ndarray` header, but does not create a copy, it creates a view only: [views versus copies](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html?highlight=view#views-vs-copies). You can see that in `numpy`:

```python
a = array(range(9)).reshape((3, 3))
print('a: ', a)
b = a.transpose()
print('\nb: ', b)
info(a)
info(b)
```
In `ulab`, you have to do it a bit differently: 

```python
a = np.array(range(9)).reshape((3, 3))
print('a: ', a)
b = a.transpose()
print('\nb: ', b)
np.ndinfo(a)
np.ndinfo(b)
```
What you have to look at is the `data pointer:` entry in the print-out. It is the same for `a`, and `b`, which means that the data that is stored in the `ndarray`s is the same. One way of creating a copy is what you have already found. But you can also specifically call the `.copy` array method

```python
a = array([...])
b = a.copy()
```

jgpeiro, 2021-02-13:
Ok, I understand the difference between view and copy and why b.transpose()+0 or b.transpose().copy() can fix my issue...
But what is the reason to needed it in my last example? I mean, why np_dot takes b "original" values instead "transposed" values?

v923z, 2021-02-13:
@jgpeiro I think I have found the error, which was in `dot`: the `transpose` re-writes the strides, which was not correctly resolved in `dot`. Could you, please, check out https://github.com/v923z/micropython-ulab/pull/319, https://github.com/v923z/micropython-ulab/tree/dot-fix, and see if your code works now. 

Thanks for bringing up the issue, by the way! 

One more question: do you think you would like to check the [test scripts](https://github.com/v923z/micropython-ulab/tree/dot-fix/tests), and see if `dot` is tested properly? Contributors get [brownie point](https://github.com/v923z/micropython-ulab/graphs/contributors), and testing is a really great way of improving the quality of a project. To be honest, we should have been able to catch this glitch much earlier, if the tests had been extensive enough. If you need help with the tests, just let me know.

jgpeiro, 2021-02-14:
@v923z Nice!. Ok, I will check the branch and review test scripts to improve them. Thanks

jgpeiro, 2021-02-15:
@v923z I verified that np.dot( a, b.transpose() ) works correctly in dot-fix branch:

```
from ulab import numpy as np

a = np.array( [
    [1,2],
    [3,4]
], dtype=np.float )

b = np.array( [
    [5,6],
    [7,8]
], dtype=np.float )

print( np.dot( a, b.transpose() ) )
```

```
Ulab Output:
array([[17.0, 23.0],
       [39.0, 53.0]], dtype=float32)

Expected:
[[17. 23.]
 [39. 53.]]
```

About improve tests, I checked the commits to transform.c but I didnt catch how it works, I mean that I can only suggest this test case (np.dot( a, b.transform() )... I also saw that "ab = np.dot(a.transpose(), b)" is already on tests/numpy/linalg.py and that was added 3 months ago... so I dont know what is the difference or how to improve. At least this case: np.dot( a, b.transform() ), but I dont know more or other cases that should be tested

v923z, 2021-02-20:
> About improve tests, I checked the commits to transform.c but I didnt catch how it works, I mean that I can only suggest this test case (np.dot( a, b.transform() )... I also saw that "ab = [...]

@jgpeiro Sorry, this issue has somehow flown off my radar. 

But to answer your question about improving the tests, you have demonstrated that the function failed, but passed the tests, which means that the tests were not adequate. So, as a bare minimum, we should definitely add this, or something similar to the test suite. Once that is done, we can close the issue.

jgpeiro, 2021-02-20:
I tried to add this code to linalg.py:
...
if use_ulab:
    a = np.array( [[1,2],[3,4]] )
    b = np.array( [[5,6],[7,8]] )
    
    print( np.dot( a, b ) )
    print( np.dot( a.transpose(), b ) )
    print( np.dot( a, b.transpose() ) )
    print( np.dot( a.transpose(), b.transpose() ) )
    ...

and linalg.py.exp:
array([[19.0, 22.0],
       [43.0, 50.0]], dtype=float32)
array([[26.0, 30.0],
       [38.0, 44.0]], dtype=float32)
array([[17.0, 23.0],
       [39.0, 53.0]], dtype=float32)
array([[23.0, 31.0],
       [34.0, 46.0]], dtype=float32)
...

but I got a permission error during push:
remote: Permission to v923z/micropython-ulab.git denied to jgpeiro.
Can you give me permissions to push this fix?

v923z, 2021-02-20:
@jgpeiro Oh, you can't do that directly. Here are the steps: 

1. create a fork from `micropython-ulab`. This will be a copy on github that you can write to.
2. clone that to your local computer
3. add the test script, and run `.build.sh`: the section [Testing](https://github.com/v923z/micropython-ulab/#testing) might be useful at this point.
4. once you are done with adding the tests to your local copy, push that back to your fork.
5. go back to github, and then create a PR against a branch (`master` in this case) in `micropython-ulab`.

https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/

By the way, irrespective of the write rights, it is never a good idea to directly push into master, even if you are allowed to (i.e., even if you do that in your own repository). Here is, why: http://waterstreetgm.org/git-why-you-should-never-commit-directly-to-master/

v923z, 2021-03-08:
@jgpeiro https://github.com/v923z/micropython-ulab/pull/344 adds the requested tests. Closing the issue.

closed 2021-03-08 by v923z