v923z/micropython-ulab #323: FYI - ESP32 port moved to CMake

jimmo, 2021-02-15:
https://forum.micropython.org/viewtopic.php?f=18&t=9820

This will mean that to build ulab with ESP32 you will need to checkout the v1.14 tag of MicroPython. Hopefully a cmake-compatible equivalent of the user c modules functionality will be added soon (to support both ESP32 and Raspberry Pi).

v923z, 2021-02-15:
@jimmo Many thanks for bringing this up, Jim! I have updated the readme, is this section OK now? https://github.com/v923z/micropython-ulab/blob/master/README.md#esp32-based-boards

ubIQio, 2021-02-15:
> Hopefully a cmake-compatible equivalent of the user c modules functionality will be added soon (to support both ESP32 and Raspberry Pi).

@jimmo thanks indeed; esp-idf is moving at a much faster pace than micropython/esp32 has traditionally,  so once user modules are supported, we can all  think about newer devices...   hopefully someone at Espressif will have the foresight to send an ESP32-S3 with vector extensions to team ulab sooner rather than later???

thalesmaoa, 2021-03-21:
I saw that you've updated some instructions. The difference from the presented ones and they way I was doing it the root directory. I will follow it again.

v923z, 2021-03-31:
I think https://github.com/v923z/micropython-ulab/pull/354, and the extended build instructions in the [readme](https://github.com/v923z/micropython-ulab#esp32-based-boards) address the issue at hand, so closing now.

closed 2021-03-31 by v923z