v923z/micropython-ulab #328: 1.7.2: ndarray_get_buffer() thinks `->array` is a python object

dhalbert, 2021-02-19:
ulab 1.7.2 still defines the routine `ndarray_get_buffer()`. This code calls `mp_get_buffer(self->array, ...)`. However, my understanding from reading `create_zeros_ones_full()` is that `self->array` is the raw array of values. It is not a Python object. `mp_get_buffer()` expects its first arg to be a Python object, so bad things happen.

https://github.com/v923z/micropython-ulab/blob/743d86487c83e42024ed508ed50499ad0a527d5d/code/ndarray.c#L1989-L1994

```c
mp_int_t ndarray_get_buffer(mp_obj_t self_in, mp_buffer_info_t *bufinfo, mp_uint_t flags) {
    ndarray_obj_t *self = MP_OBJ_TO_PTR(self_in);
    // buffer_p.get_buffer() returns zero for success, while mp_get_buffer returns true for success
    return !mp_get_buffer(self->array, bufinfo, flags);
}
```

I had some code similar to this in CircuitPython:
```python
a = ulab.zeros(10, dtype=ulab.uint16)
some_bleio_characteristic.value = a
```
`_bleio.Characteristic.value` expects the value coming in to obey buffer protocol, and ends up called `mp_get_buffer()` on the `ndarray` itself. That in turn calls `ndarray_get_buffer()`, which tries to call `mp_get_buffer()` on the `array` itself, and things go wrong.

`ndarray_get_buffer()` seems to have disappeared in 2.x `ulab`, so I'm not sure how to fix 1.7.2. Is there something new to backport to 1.7.x? Thanks.

v923z, 2021-02-19:
> ulab 1.7.2 still defines the routine `ndarray_get_buffer()`. This code calls `mp_get_buffer(self->array, ...)`. However, my understanding from reading `create_zeros_ones_full()` is that `self->array` is the raw array of values. It is not a Python object. 

That's right. `mp_get_buffer()` is leftover of the original code, when the `ndarray` was derived from `micropython`'s own `array` object, which I bid farewell to with version 1.0. I somehow overlooked that. Sorry for that, and thanks for pointing out the problem!

> `_bleio.Characteristic.value` expects the value coming in to obey buffer protocol, and calls `mp_get_buffer()` on the `ndarray`. That in turn calls `ndarray_get_buffer()`, which tries to call `mp_get_buffer()` on the `array` itself, and things go wrong.

> `ndarray_get_buffer()` seems to have disappeared in 2.x `ulab`, so I'm not sure how to fix 1.7.2. Is there something new to backport to 1.7.x? Thanks.

I think @jepler has already sorted this out in https://github.com/v923z/micropython-ulab/pull/329. The long-term question is, however, whether `ndarray_get_buffer` is needed at all. I believe, the explicit `frombuffer` function should be able to handle everything that `ndarray_get_buffer` can do, but perhaps, you still want to be able to just toss a buffer into the constructor.

dhalbert, 2021-02-19:
> I believe, the explicit `frombuffer` function should be able to handle everything that `ndarray_get_buffer` can do, but perhaps, you still want to be able to just toss a buffer into the constructor.

I saw `frombuffer` but I thought that was doing the reverse of what I wanted, which is to, say, copy an `ndarray` to a `bytearray`. I was look for `tobuffer`. Maybe I'm confused.

v923z, 2021-02-20:
> > I believe, the explicit `frombuffer` function should be able to handle everything that `ndarray_get_buffer` can do, but perhaps, you still want to be able to just toss a buffer into the constructor.
> 
> I saw `frombuffer` but I thought that was doing the reverse of what I wanted, which is to, say, copy an `ndarray` to a `bytearray`. I was looking for `tobuffer`. Maybe I'm confused. And what I need is all internal, not a Python API.

I don't know, whether `tobytes` would do: https://github.com/v923z/micropython-ulab/blob/f4604d208b9d92d4fa8869b74a54ebb71a6ed101/code/ndarray.c#L1549-L1558

`mp_obj_new_bytearray_by_ref` should produce a buffer that you can use, but I am not 100% sure. 

One more note on this subject: as the comment states, the `tobytes` function is not `numpy`-compatible, because it doesn't produce a copy. I opted for this solution to save RAM, but I am not against changing this. However, I would still like to have a function/method that just lays of the pointer of the `ndarray` bare, so that you get something that you can directly pipe into a DAC, I2S device, etc. I haven't yet found the corresponding `numpy` function.

Perhaps, the solution is not in `numpy` at all, and we are barking up the wrong tree. I am really close to finishing a complete new module, independent of `numpy`, but compatible with the `ndarray` object that would make extensions extremely flexible. When all dust settles down, we could delegate such issues to that module. My vision is that one should be able to extend `ulab` without having to add a single line to the `ulab` core.

dhalbert, 2021-02-20:
Fixed by #329 for my original problem. IO will close, but @v923z, if you want to create a new issue for the tail of the discussion, sounds like a good idea.

PR #329 (2021-02-19): ndarray: Fix memoryview(ulab.array(...))
For now this only handles the 1D case.  In theory it would work for any dense array, however, I found that `ndarray_is_dense` didn't behave for me so I implemented this instead.

Add a test.  Before the change, this test would segfault.

Closes #328.
commit 7aeb73a1 (2021-02-19): ndarray: Fix memoryview(ulab.array(...))

For now this only handles the 1D case.  In theory it would work for
any dense array, however, I found that ndarray_is_dense didn't behave
for me so I implemented this instead.

Add a test.  Before the change, this test would segfault.

Closes #328.
closed 2021-02-20 by dhalbert