v923z/micropython-ulab #331: [BUG] np.eye(

jepler, 2021-02-19:
**Describe the bug**
np.eye(dtype=bool) returns all elements `False`.

Occurs at:
 - micropython-ulab legacy branch at 8d156616bf0ad1993e76357612623f59fdc9153c
 - micropython unix port as built by ./build.sh at v1.13-268-gf7aafc062

```python
MicroPython v1.13-268-gf7aafc062 on 2021-02-19; linux version
>>> import ulab as np
>>> np.__version__
'1.7.2-2D-numpy'
```

**To Reproduce**

```python
>>> np.eye(3, dtype=np.bool)
array([[False, False, False],
       [False, False, False],
       [False, False, False]], dtype=bool)
```

**Expected behavior**
Standard Python/numpy prints
```
>>> import numpy as np
>>> np.eye(3, dtype=np.bool)
array([[ True, False, False],
       [False,  True, False],
       [False, False,  True]])
```

v923z, 2021-02-20:
@jepler https://github.com/v923z/micropython-ulab/pull/332 is a fix for this, and a couple of related bugs. I might have to do that for `linspace`, too.

v923z, 2021-03-08:
@jepler https://github.com/v923z/micropython-ulab/pull/343 and https://github.com/v923z/micropython-ulab/pull/345 implements the fix on both relevant branches, closing the issue.

closed 2021-03-08 by v923z