v923z/micropython-ulab #335: [FEATURE REQUEST] support buffer protocol?

v923z, 2021-02-21:
@dhalbert, @jepler 

`ulab` version 2.0 removed the `ndarray_get_buffer()` C function, and with that the support for buffer protocol. The rationale was that the explicit `frombuffer/tobytes` `python` functions should handle interaction with buffers, and at the time, `ndarray_get_buffer()` appeared to be duplication of functionality. 

However, based on the comments in https://github.com/v923z/micropython-ulab/issues/328, it might be reasonable to re-introduce the function, and control inclusion via the [ulab.h](https://github.com/v923z/micropython-ulab/blob/master/code/ulab.h) header file, or constants passed on the command line to the pre-processor.

jepler, 2021-02-21:
I think that you need this function, as it is used in two ways:
 * from Python code, it makes a regression (compared to legacy branch and compared to standard python) in which `memoryview(ndarray)` will not work, unless there's something I'm overlooking.
 * internally, code which wants to treat an ndarray as a bunch of memory will use the `mp_get_buffer` and `mp_get_buffer_raise` functions (this is what `memoryview(ndarray)` does, but we do it other places)

v923z, 2021-02-21:
I think these are quite pressing reasons, so I will add then this to the master branch.

v923z, 2021-02-21:
This issue is fixed in https://github.com/v923z/micropython-ulab/pull/338, closing now.

closed 2021-02-21 by v923z