v923z/micropython-ulab #336: [FEATURE REQUEST] backport `byteswap`

v923z, 2021-02-21:
@dhalbert, @jepler Do you want me to backport this function? Here is the broader context: https://github.com/v923z/micropython-ulab/issues/306

jepler, 2021-02-21:
If the original requestor of the feature was using CircuitPython, then probably yes.  Otherwise, not necessarily.

v923z, 2021-02-21:
> If the original requestor of the feature was using CircuitPython

No, he wasn't. However, you are guaranteed to run into the endianness problem, if you try to pipe data from a peripheral device into a numerical array. On the other hand, at one point `circuitpython` is also going to switch to the `master` branch, so if you say that this is not a burning issue for you, I am not going to complain.

v923z, 2021-03-08:
Closing the issue.  A new issue should be opened, if the functionality is required in `circuitpython`.

closed 2021-03-08 by v923z