v923z/micropython-ulab #349: [BUG]

thalesmaoa, 2021-03-15:
**Describe the bug**
Can't compile ulab using README instructions. It seems that something is missing.

```python
$ make submodules & make all
[1] 12329
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ./ulab/code
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ./ulab/code
Building with ESP IDF v3
Building with ESP IDF v3
Updating submodules: lib/berkeley-db-1.xx
CC ./ulab/code/scipy/optimize/optimize.c
./ulab/code/scipy/optimize/optimize.c: In function 'optimize_bisect':
./ulab/code/scipy/optimize/optimize.c:64:11: error: 'MP_QSTR_xtol' undeclared (first use in this function)
         { MP_QSTR_xtol, MP_ARG_KW_ONLY | MP_ARG_OBJ, {.u_rom_obj = MP_ROM_PTR(&xtolerance)} },
           ^
```

**To Reproduce**
Follow ESP32 compilation instructions from README

**Expected behavior**
Compile elf file.

v923z, 2021-03-15:
@thalesmaoa Can you run a `make clean` before anything else?

thalesmaoa, 2021-03-15:
Yes. I did. Result is the same. I also made in a fresh new folder.

v923z, 2021-03-15:
Can you compile for the `unix` port, or is this issue related to the ESP32 only?

v923z, 2021-03-15:
Oh, I might know what the problem is: you are still pinning the `micropython` version to 1.14, right? We have added the `cmake` fragment: https://github.com/v923z/micropython-ulab/blob/master/code/usermod.cmake. Can you unpin the `micropython` version, and try to compile again (starting with `make clean`)?

thalesmaoa, 2021-03-15:
Sure. I'm doing a fresh install here. Will let you know soon.

v923z, 2021-03-21:
@thalesmaoa Sorry, I haven't seen your edits. You should always open a new section in the thread, otherwise, I don't get notifications. 

Have you read https://github.com/v923z/micropython-ulab/issues/323? I am not set up for ESP32 at the moment, but I will try to install the dependencies, and see what can be done.

What happens, if you delete the `cmake` fragment https://github.com/v923z/micropython-ulab/blob/master/code/usermod.cmake?

v923z, 2021-03-21:
@thalesmaoa The issue has automatically been closed by https://github.com/v923z/micropython-ulab/pull/351. I could compile the firmware with the updated instructions. If you still have difficulties, re-open the issue!

closed 2021-03-21 by v923z

v923z, 2021-03-23:
@thalesmaoa Here is a script that should compile for the ESP32 port: https://github.com/v923z/micropython-ulab/tree/master/build 
simply download the file (`ulab` will be cloned automatically), and run the script.

thalesmaoa, 2021-03-23:
Perfect. Thanks.
-
Thales Alexandre Carvalho Maia