v923z/micropython-ulab #365: [BUG]

nonix, 2021-04-18:
**Describe the bug**
Just a small README.md correction.
In section [ESP32-based-boards](https://github.com/v923z/micropython-ulab#esp32-based-boards) 
"Finally, build the firmware:"

- ESPDIF env. variable needs to be set to ../../esp32/esp-idf
- mpy-cross needs to build first:
     cd ../../mpy-cross
     make
     cd -

```python
from ulab import numpy as np
print(np.__version__)
```

**To Reproduce**
I've just followed the steps ...

**Expected behavior**
to work

**Additional context**

v923z, 2021-04-18:
> **Describe the bug**
> Just a small README.md correction.
> In section [ESP32-based-boards](https://github.com/v923z/micropython-ulab#esp32-based-boards)
> "Finally, build the firmware:"
> 
>     * ESPDIF env. variable needs to be set to ../../esp32/esp-idf

True, thanks for pointing that out!

> 
>     * mpy-cross needs to build first:

But that is the very first step in that section, after cloning the repositories. 

>       cd ../../mpy-cross
>       make
>       cd -

By the way, you can simply take the build script https://github.com/v923z/micropython-ulab/blob/master/build/esp32.sh.

v923z, 2021-04-28:
The corresponding section in the README has been overhauled in https://github.com/v923z/micropython-ulab/pull/376, so closing the issue now.

closed 2021-04-28 by v923z