v923z/micropython-ulab #378: Implementation of QR factorization

vikas-udupa, 2021-05-05:
Hello @v923z. I noticed that in qr branch, you are currently working on the implementation of QR factorization, which I feel would be very useful. I have a couple of questions regarding the same. 

1. In order to implement the factorization, are you implementing the Householder algorithm or the Gram-Schmidt algorithm?
2. What stage of the implementation are you currently at and by when are you expecting to complete it? I have a working implementation of Householder QR in micropython ulab that I am happy to share if it may be useful.

Thanks!

v923z, 2021-05-05:
> Hello @v923z. I noticed that in qr branch, you are currently working on the implementation of QR factorization, which I feel would be very useful. I have a couple of questions regarding the [...]

Gram-Schmidt is numerically ill-conditioned. Householder could be OK, but I actually started to work with Givens rotations.  

>     2. What stage of the implementation are you currently at and by when are you expecting to complete it? I have a working implementation of Householder QR in micropython ulab that I am happy to share if it may be useful.

I think the implementation is complete, but there is a sign error somewhere, which I haven't yet found. I'll try to fix it over the weekend at the latest. However, if you have something that's already working, that would definitely be useful.

v923z, 2021-05-11:
@vikas-udupa I think the code should be OK now, at least for the `mode='complete'` case: https://github.com/v923z/micropython-ulab/pull/371 
If you have comments, then in the thread of the PR, please!

v923z, 2021-05-22:
@vikas-udupa Do you think you could look at https://github.com/v923z/micropython-ulab/pull/371?

v923z, 2021-07-22:
Implemented in https://github.com/v923z/micropython-ulab/pull/427

closed 2021-07-22 by v923z