v923z/micropython-ulab #382: [BUG]ulab does not compile with latest circuitpython

v923z, 2021-05-09:
@jepler The CI (and consequently, `circuitpython` build) fails with the latest `circuitpython`: https://github.com/v923z/micropython-ulab/runs/2537755096

I don't quite see at the moment what the problem is, but I am pretty sure it is caused by a modification in the `circuitpython` code: I re-ran the jobs with https://github.com/v923z/micropython-ulab/commit/a61cd88f245bddc49f0eda8158de2abd080b4dd3, which worked two days ago, but even that fell through. 

You could definitely merge your changes, if you can patch it up. I have worked on the documentation only, so I have merged my commits, even with the failing CI.

v923z, 2021-05-10:
@jepler Many thanks!

v923z, 2021-05-10:
@jepler Jeff, sorry, I have to re-open this: https://github.com/v923z/micropython-ulab/pull/371 fails on the push: https://github.com/v923z/micropython-ulab/pull/371/checks?check_run_id=2548339155

I am really confused now, given that push and pull are treated on the same footing in the workflow...

jepler, 2021-05-10:
I think that if you merge master into your pr branch that it will pass on push. "Push" runs the branch exactly as is, and the branch itself doesn't have the fix for this issue. The pull request build creates a merge commit and builds that, so it includes the fix.

v923z, 2021-05-11:
> I think that if you merge master into your pr branch that it will pass on push. "Push" runs the branch exactly as is, and the branch itself doesn't have the fix for this issue. 

Oh, sorry, that makes sense!

v923z, 2021-05-11:
@jepler I have rebased the `qr` branch, the tests pass now. Sorry for the noise!